arduino/Arduino #4051: [Library Manager] How long should it take for a new release to appear?

ghlawrence2000, 2015-10-30:
Hello, I am new to the library manager thing having only submitted my first release of my first library back in August. I have since made some updates to the library and have created a new Github release.

My library is UTFT_SdRaw and I now have version 1.2.0 and version 1.2.2, V1.2.2 was created 15 hours ago but has still not shown up in Library Manager yet, is there anything else I need to do?

The Github address is this :- https://github.com/ghlawrence2000/UTFT_SdRaw/releases/tag/v1.2.2.

Sorry if I am being impatient, or asking newbie questions!

Regards,

Graham

Chris--A, 2015-10-30:
Delete your release, and its tag, then re-release and it will include the last commit: https://github.com/ghlawrence2000/UTFT_SdRaw/commit/38f15e622b7fd7f3ab5a0132a27f0f30417d768b

When done, wait an hour and check again.

ghlawrence2000, 2015-10-30:
Ok, but how come I have seen other libraries within Library Manager with multiple Versions?

Chris--A, 2015-10-30:
You can have multiple releases, however the latest release contains 1.2.0 in library.properties, which is the same as the first release, not 1.2.2

ghlawrence2000, 2015-10-30:
So you are saying I need to delete release v1.2.2 and re-create it?

Chris--A, 2015-10-30:
Yeah, but you will probably have to delete the tag too, not just the release.

ghlawrence2000, 2015-10-30:
Ok thank you, I will give it a go.......

Graham

ghlawrence2000, 2015-10-30:
I cannot see how to delete a release, but having checked, the library.properties does say version 1.2.2 in the zip file of the release, what else am I doing wrong?

Chris--A, 2015-10-30:
I checked that too, strange, it clearly says the commit was added to master since the release.
Seems strange, I do not think it should be included.

I saw you deleted the release, can't hurt retrying it anyway. If it does not fix the problem, the arduino folk will be online in a few hours.

ghlawrence2000, 2015-10-30:
Thanks for your help, I just figured how to delete. I will try it again.

Regards

Chris--A, 2015-10-30:
Its in the json file now.

ghlawrence2000, 2015-10-30:
Thanks. But that is strange, all I did was delete the release and re-submit it with the SAME zip from yesterday.... But anyway, if it is fixed now that is great!

Regards,

Graham

NicoHood, 2015-10-30:
If your question is not answered fully in the FAQ, but you now know more about it, maybe someone can add it here?
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

closed 2015-10-30 by ffissore

ghlawrence2000, 2015-10-30:
Federico, you were a bit quick off the mark closing this issue................. it still has not appeared in library manager yet...........