arduino/Arduino #4052: K12 environment and Windows Group Policy

pamccall, 2015-10-30:
We are using this in Career Tech in our high schools and we use Windows Group Policy to prevent student users from unintentionally or intentionally deleting files from PCs that would prevent the PC from working.  After install this software recently and making sure the C:\Program Files (x86)\Arduino path was open to be accessed by students the program still will not launch.  When you double click the icon the screen flashes but the program doesn't open.  It seems like the software may be going somewhere else other than this path when launched that may be blocked but I can't tell where in the OS that is happening?  Anyone have any insight into this that can help?

NicoHood, 2015-10-30:
%appdata%/Arduino15 or something like that.
In the home folder is also a sketchbook folder named Arduino where you can place libraries and save sketches. This also should be unlocked.

ffissore, 2015-10-30:
Indeed.
That path is used to store preferences, boards support installed through boards manager, and libraries installed through library manager.
If you want the IDE to use its installation path, you need to create a `portable` folder inside it.

However, the IDE needs a folder where to store its sketchbook, which by default on Windows is the `Arduino` in `My Documents` folder.
This be changed via File > Preferences

NicoHood, 2015-10-30:
I heard about the portable folder never before. Is this somewhere documented? Maybe you should add this to the .zip files by default or as Setting? Thats sounds like a cool feature on school PCs which are mostly windows and blocked.

pamccall, 2015-10-30:
I have tried adding to Security Settings in Group Policies the following:
 %AllUsersProfile%\Arduino 15
%SystemDrive%\Users\Default\AppData\Roaming\Arduino15
%ProgramFiles%(x86)\Arduino 
None of these helped.  In addition we opened the Program and went to File and Preferences and changed the Sketchbook path to a mapped drive for the students where they can save.  Students can not save to My Documents on the local machine.  We force them to save to a network drive so that we can backup their work.  When we try to open the program we still only get a brief flash and then the program never opens.  I am at a loss as to what else we should try.  Any ideas?

ffissore, 2015-10-30:
Have you created a `portable` folder, as I mentioned in https://github.com/arduino/Arduino/issues/4052#issuecomment-152519782 ?

pamccall, 2015-10-30:
Yes, I have created that folder but still same result

pamccall, 2015-10-30:
> Have you created a portable folder, as I mentioned in #4052 (comment)

Yes, I have created that folder.  Still same result.

-- 
Pam McCall
IT Systems Administrator
HCPS Technology

ffissore, 2015-10-30:
Try running the IDE from the command line with `arduino_debug.exe`. It will print an error message. Paste it here

pamccall, 2015-11-02:
Students do not have access to command lines - restricted.  Program opens
file as an administrator so I am not sure running the debug from
there would be helpful.  Any other ideas?

-- 
Pam McCall
IT Systems Administrator
HCPS Technology

ffissore, 2015-11-02:
Create a shortcut to `arduino_debug.exe` and prepend `C:\Windows\System32\cmd.exe /k ` (final space) to `Target`

ffissore, 2015-11-02:
Then use that shortcut to start the IDE. An error message will be printed. Please paste it here

pamccall, 2015-11-02:
Thanks!  We tried another route.  Another laptop has this software but the
previous version 16.1 and it works as a student on this version.  I just
downloaded and install this version after uninstalling the newest and
verified the older version works.  That tells me something in the source
code changed and for us that meant anyone who was a non-admin local use on
the PC could not use the new software.

-- 
Pam McCall
IT Systems Administrator
HCPS Technology

ffissore, 2015-11-02:
Good for you, although that doesn't help us finding the problem, which is likely to remain unresolved, unless someone else comes with more details

pamccall, 2015-11-02:
I will try again later today to install on a different laptop and still get
the debugging info to you.

-- 
Pam McCall
IT Systems Administrator
HCPS Technology

agdl, 2016-07-12:
I'm closing this for lack of feedback. Please reopen this if needed

closed 2016-07-12 by agdl