arduino/Arduino #4055: .tmp/core\core.a: No such file or directory

mpflaga, 2015-10-30:
When using DigiSpark Board from Broad Manager "Hourly Build 2015/10/30 03:43" fails to compile provided examples. However, "Hourly Build 2015/08/28 11:09" does compile successfully. 

This smells like a more chronic issue than related to a custom board.

Below is failed compile output from: Then farther below is a successful compile from prior hourly build. A diff between them shows some significant differences.
> Hourly Build 2015/10/30 03:43
>
> C:\projects\Arduino\arduino-nightly.2015_10_30\arduino-builder -dump-prefs -logger=machine -hardware "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware" -hardware "C:\Users\mflaga\AppData\Local\Arduino15\packages" -tools "C:\projects\Arduino\arduino-nightly.2015_10_30\tools-builder" -tools "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr" -tools "C:\Users\mflaga\AppData\Local\Arduino15\packages" -built-in-libraries "C:\projects\Arduino\arduino-nightly.2015_10_30\libraries" -libraries "C:\Users\mflaga\Documents\Arduino\libraries" -fqbn=digistump:avr:digispark-tiny -ide-version=10606 -build-path "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB\examples\DigisparkRGB\DigisparkRGB.ino"
> C:\projects\Arduino\arduino-nightly.2015_10_30\arduino-builder -compile -logger=machine -hardware "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware" -hardware "C:\Users\mflaga\AppData\Local\Arduino15\packages" -tools "C:\projects\Arduino\arduino-nightly.2015_10_30\tools-builder" -tools "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr" -tools "C:\Users\mflaga\AppData\Local\Arduino15\packages" -built-in-libraries "C:\projects\Arduino\arduino-nightly.2015_10_30\libraries" -libraries "C:\Users\mflaga\Documents\Arduino\libraries" -fqbn=digistump:avr:digispark-tiny -ide-version=10606 -build-path "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB\examples\DigisparkRGB\DigisparkRGB.ino"
> WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
> WARNING: Spurious .git folder in 'Adafruit ILI9340' library
> WARNING: Spurious .git folder in 'LinkedList' library
> WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR   "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR   "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR   "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR   "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR   "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB\DigisparkRGB.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -w -x c++ -M -MG -MP -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR   "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++"  -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR         "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\tools-builder\ctags\5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\preproc\ctags_target.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\tools-builder\ctags\5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\preproc\ctags_target.cpp"
> "C:\projects\Arduino\arduino-nightly.2015_10_30\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10606 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR  "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\tiny" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\digispark" "-IC:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\libraries\DigisparkRGB" "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp" -o "C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\sketch\DigisparkRGB.ino.cpp.o"
> Using previously compiled file: C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\libraries\DigisparkRGB\DigisparkRGB.cpp.o
> Using previously compiled file: C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\core\pins_arduino.c.o
> Using previously compiled file: C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\core\WInterrupts.c.o
> Using previously compiled file: C:\Users\mflaga\AppData\Local\Temp\build634669761f7dbb6ce1dba00e5ab8798d.tmp\core\pins_arduino.c.o
[... cut ...] pp.elf C:\Users\mflaga\AppData\Local\Temp\build8446471256195278974.tmp/DigisparkRGB.cpp.eep 
> C:\projects\Arduino\arduino-nightly.2015_08_29\hardware\tools\avr/bin/avr-objcopy -O ihex -R .eeprom C:\Users\mflaga\AppData\Local\Temp\build8446471256195278974.tmp/DigisparkRGB.cpp.elf C:\Users\mflaga\AppData\Local\Temp\build8446471256195278974.tmp/DigisparkRGB.cpp.hex 
> 
> Sketch uses 872 bytes (14%) of program storage space. Maximum is 6,012 bytes.
> Global variables use 23 bytes of dynamic memory.

matthijskooijman, 2015-10-31:
It seems that the recipes in platform.txt might not be entirely consistent, though it's weird it does work with the older version. Could you attach the platform.txt file from the Digispark core you are using?

PaulStoffregen, 2015-10-31:
I had to update my platform.txt to deal with the (much needed) change where core.a is built in a "core" subdir.  Pretty simple to do.  My platform.txt disables rewriting.

mpflaga, 2015-10-31:
The compile output indicates C:\Users\mflaga\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.5\platform.txt, I don't have write permission to attach. so pasting below.

I am not knowledgeable enough to know what Paul disabled or how to in the below DigiSpark platform.

     # Arduino AVR Core and platform.
     # ------------------------------
     
     # For more info:
     # https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification
     
     name=Digistump AVR Boards
     version=1.5.4
     
     # AVR compile variables
     # --------------------- 
     
     # Default "compiler.path" is correct, change only if you want to overidde the initial value
     compiler.path={runtime.tools.avr-gcc.path}/bin/
     compiler.c.cmd=avr-gcc
     compiler.c.flags=-c -g -Os -w -ffunction-sections -fdata-sections -MMD
     compiler.c.elf.flags=-Os -Wl,--gc-sections
     compiler.c.elf.cmd=avr-gcc
     compiler.S.flags=-c -g -x assembler-with-cpp
     compiler.cpp.cmd=avr-g++
     compiler.cpp.flags=-c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD
     compiler.ar.cmd=avr-ar
     compiler.ar.flags=rcs
     compiler.objcopy.cmd=avr-objcopy
     compiler.objcopy.eep.flags=-O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0
     compiler.elf2hex.flags=-O ihex -R .eeprom
     compiler.elf2hex.cmd=avr-objcopy
     compiler.ldflags=
     compiler.size.cmd=avr-size
     # this can be overriden in boards.txt
     build.extra_flags=
     
     # AVR compile patterns
     # --------------------
     
     ## Compile c files
     recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
     
     ## Compile c++ files
     recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
     
     ## Compile S files
     recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.S.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
     
     ## Create archives
     recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}"
     
     ## Combine gc-sections, archives, and objects
     recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm
     
     ## Create eeprom
     recipe.objcopy.eep.pattern="{compiler.path}{compiler.objcopy.cmd}" {compiler.objcopy.eep.flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.eep"
     
     ## Create hex
     recipe.objcopy.hex.pattern="{compiler.path}{compiler.elf2hex.cmd}" {compiler.elf2hex.flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.hex"
     
     ## Compute size
     recipe.size.pattern="{compiler.path}{compiler.size.cmd}" -A "{build.path}/{build.project_name}.elf"
     recipe.size.regex=^(?:\.text|\.data|\.bootloader)\s+([0-9]+).*
     recipe.size.regex.data=^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
     recipe.size.regex.eeprom=^(?:\.eeprom)\s+([0-9]+).*
     
     
     # AVR Uploader/Programmers tools
     # ------------------------------
     
     #tools.micronucleus.cmd.path={runtime.ide.path}/hardware/digistump/avr/tools/avrdude
     #tools.micronucleus.cmd.path.linux={runtime.ide.path}/hardware/digistump/avr/tools/avrdude
     tools.micronucleus.cmd.path={runtime.tools.micronucleus.path}/launcher
     
     tools.micronucleus.upload.params.verbose=
     tools.micronucleus.upload.params.quiet=
     #tools.micronucleus.upload.pattern="{cmd.path}" --run --timeout 60 "{build.path}/{build.project_name}.hex"
     tools.micronucleus.upload.pattern="{cmd.path}" -cdigispark --timeout 60 -Uflash:w:{build.path}/{build.project_name}.hex:i
      
     # USB Default Flags
     # Default blank usb manufacturer will be filled it at compile time
     # - from numeric vendor ID, set to Unknown otherwise
     build.usb_manufacturer=
     build.usb_flags=

ffissore, 2015-11-02:
This issue was moved to arduino/arduino-builder#42

closed 2015-11-02 by ffissore