arduino/Arduino #4056: Ambiguous SPI documentation

gwhughes, 2015-10-30:
Under "Pin Mapping" on https://www.arduino.cc/en/Main/ArduinoBoardMega2560, it reads :SPI: 50 (MISO), 51 (MOSI), 52 (SCK), 53 (SS). These pins support SPI communication using the SPI library. 

Whereas under "Input and Output" on https://www.arduino.cc/en/Main/arduinoBoardMega , it reads: SPI: 50 (MISO), 51 (MOSI), 52 (SCK), 53 (SS). These pins support SPI communication, which, although provided by the underlying hardware, is not currently included in the Arduino language.

Which is correct?

PaulStoffregen, 2015-10-30:
Both should say "These pins support SPI communication using the SPI library."

agdl, 2015-11-26:
Now the documentation is correct. I'm closing this

closed 2015-11-26 by agdl