arduino/Arduino #4057: double-click to open sketch does not

i-make-robots, 2015-10-30:
windows 8.
double click sketch/sketch.ino
"sketch folder missing.  will create new default sketch folder if required, then stop talking about oneself in the third person."
click ok.
app quits.
....
try again, same results
....
click arduino icon.
app gives error message again.
click ok.
app opens two copies of sketch/sketch.ino

ffissore, 2015-11-02:
Which version of the IDE are you using?

i-make-robots, 2015-11-02:
OSX 10.10.1
Arduino 1.6.4

Jed-SWSDT, 2015-11-19:
I can't duplicate that exact issue with Arduino 1.6.6 on OSX 10.10.5. But I do see a similar issue. If Arduino isn't running and I double-click a sketch it opens Arduino, but loads the last sketch I was working on, not that one I double-clicked on. 

Was able to get it to move it to a new folder. Double-click on sketch outside of a folder, opens Arduino to the WRONG sketch. Double-click on the sketch outside of the folder again and a different dialog box pops up for me:

"The file "xxx.ino" needs to be inside a sketch folder named "xxx". Create this folder, move the file, and continue?"

Was able to do it, at least on the desktop.

ffissore, 2015-11-20:
Working on it

ffissore, 2015-11-20:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit 7e9f19c4 (2015-11-20): MacOSX: faster startup, fixed double clicking on a .ino file. Fixes #4057
closed 2015-11-20 by ffissore