arduino/Arduino #4058: two separate windows.  why not two separate serial monitors?

i-make-robots, 2015-10-30:
Because I have several arduinos on different ports talking at once and I want to monitor all of them, please.

NicoHood, 2015-10-31:
You can open a new Arduino instance to work around this. However I also like to have this feature inside a single instance. I normally use another Arduino/FTDI to analyze the debug output via Serial1.

ffissore, 2015-11-02:
Indeed: you'd better use some external serial tool like putty. Having different serial monitors would mean bonding each to each sketch window, possibly leading to opening two serial monitors on the same port and having windows that pops up while other disappears.

closed 2015-11-02 by ffissore

PaulStoffregen, 2015-11-02:
External terminal emulators are not a good solution, because they have no awareness of when Arduino needs to open the port.  The Arduino Serial Monitor automatically closes the port when you upload, so the IDE or AVRDUDE open it.

I'm not necessarily saying this feature should be implemented... I'm just saying the idea of using external terminals like putty or coolterm is problematic.

i-make-robots, 2015-11-02:
There is a middle ground. Have windows request status of serial connections from a manager. Then each window would visually indicate ports already in use by other windows.  That's how I handle it in non-arduino programs.