arduino/Arduino #4067: Templates not supported in sketch

xEtherealx, 2015-11-02:
The following sketch fails to compile:

    #include "Arduino.h"
    template <typename T> T test (T a) { return a;}
    void setup () { }
    void loop () { }

With `error: 'T' does not name a type`

PaulStoffregen, 2015-11-02:
Everyone should test with the latest hourly build before reporting any issue!

[image]

xEtherealx, 2015-11-02:
Ah thanks, that works!  But the following does not:

    #include "Arduino.h"
    template <typename T>
    T test (T a) { return a;}
    void setup () { }
    void loop () { }

matthijskooijman, 2015-11-02:
Hm, that's weird. The preprocessed source file looks like this:
```
#include "Arduino.h"
template <typename T>
T test (T a) { return a;}
void setup () { }
void loop () { }
```

It seems the prototype is generated correctly, but the `template<typename T>` line is removed from the original function. Or, perhaps the prototype is generated after the template line. It seems that is more likely, since if I move the template function to the end of the file, the preprocessed looks like this:

```
#include <Arduino.h>
#line 1
#line 1 "/tmp/arduino_9333c6f8a78b8b32a72543fe9017636c/sketch_nov02a.ino"
#include "Arduino.h"

void setup();
void loop();
T test (T a) ;
#line 3
void setup () { }
void loop () { }
template <typename T>
T test (T a) { return a;}
```
@ffissore, I guess ctags doesn't completely understand templates after all, then?

matthijskooijman, 2015-11-02:
ctags output looks like this:
```
setup   /tmp/build9333c6f8a78b8b32a72543fe9017636c.tmp/preproc/ctags_target.cpp /^void setup () { }$/;" kind:function   line:4  signature:()    returntype:void
loop    /tmp/build9333c6f8a78b8b32a72543fe9017636c.tmp/preproc/ctags_target.cpp /^void loop () { }$/;"  kind:function   line:5  signature:()    returntype:void
test    /tmp/build9333c6f8a78b8b32a72543fe9017636c.tmp/preproc/ctags_target.cpp /^T test (T a) { return a;}$/;" kind:function   line:7  signature:(T a) returntype:templateT
```

and for the working version with the template on a single line:

```
setup   /tmp/build9333c6f8a78b8b32a72543fe9017636c.tmp/preproc/ctags_target.cpp /^void setup () { }$/;" kind:function   line:4  signature:()    returntype:void
loop    /tmp/build9333c6f8a78b8b32a72543fe9017636c.tmp/preproc/ctags_target.cpp /^void loop () { }$/;"  kind:function   line:5  signature:()    returntype:void
test    /tmp/build9333c6f8a78b8b32a72543fe9017636c.tmp/preproc/ctags_target.cpp /^template <typename T> T test (T a) { return a;}$/;"   kind:function   line:6signature:(T a)  returntype:templateT
```

The gist of the ctags output looks identical, so I guess that for template functions, the prototype is built from the code line, instead of the function signature like normally? @ffissore, if this is so, perhaps we should drop that and just not generate prototypes for template functions?

Chris--A, 2015-11-02:
I'm not sure its a valid solution having simple functions generated, and basically everything else not. Maybe its time to stop hiding the basics of C++ and not generate any prototypes at all. It'll break old code, however the new arduino-builder has already broken previously working prototypes.

I have a good list here: https://github.com/arduino/arduino-builder/issues/30

I do not think it's going to be much of a problem, as the forum is already & still going to be bombarded with questions asking why some functions do not work.

PaulStoffregen, 2015-11-02:
> Maybe its time to ... not generate any prototypes at all

No.  Massively breaking compatibility is not an option!

Chris--A, 2015-11-02:
Well as it stands, that is what is happening. This really needs to be sorted out, I just do not think disabling prototype generation for a majority of function signatures is an option either. If were going to tell people to generate their own for templates, bound return types, functions with optional parameters, (list goes on)... then we might as well disable it all (KISS, we don't need two rules for the same thing)!  I know... just pointing out the obvious. :smile:

ffissore, 2015-11-03:
@matthijskooijman as you found out, ctags output includes one code line, not "function".
Please @xEtherealx stick on writing function on line line only.

I will not disable template generation as it works well when functions are written all on one line. If "one line" is not an option, writing the prototype yourself is the workaround.

This was one of the main goal of arduino-builder: do whatever you can to make things just work and provide options when they don't work. Previous prototype generation failed miserably, leaving users the only option to stop using the IDE: templates are an example, see #472
If you don't wish to trigger prototype generation, write them: arduino-builder will just skip all the already available prototypes.

As for https://github.com/arduino/arduino-builder/issues/30, some work is already available (see "functionpointer" branch): I just don't yet feel confident enough with it.

ffissore, 2015-11-03:
This issue was moved to arduino/arduino-builder#43

closed 2015-11-03 by ffissore