arduino/Arduino #4071: GsmWebClient example for GSM shield v2 does not work since v.1.5.7

SerxioSierra, 2015-11-03:
I have noticed that the GsmWebClient example 

http://arduino.cc/en/Tutorial/GSMExamplesWebClient

stopped working properly since Arduino IDE v.1.5.7. I have been doing tests and the problem is that the logo is not shown properly in the serial monitor window. Sometimes appears a part of the logo (some characteres) and noticed that the horizontal scroll bar is moving constantly to the right, so the logo is not shown and not completed. The disconnecting message is not shown, it seems to be in a loop writing spaces on the log.

Tests has been done using the newest Arduino GSM Shield v2 and different IDE versions including the lastest today v.1.6.6. All cases the Windows versions (in fact 8.1) and using 9600 bauds for the connection.

However using Arduino v.1.5.6-r2 that example works fine. So maybe the reason is in the GSM libraries included since v1.5.7.

Two SIMS used for the tests and in all cases with GPRS connectivity available.

Best Regads

SerxioSierra, 2015-11-14:
As I he explained in one thread in Arduino.org

http://labs.arduino.org/tiki-view_forum_thread.php?comments_parentId=988&topics_offset=2

the origin of the mentioned problem was in that I had been using a compatible board DCcEle DCcduino UNO and not the original Arduino UNO R3 board. I tested with the original board and the GsmWebClient example worked fine with Arduino IDE 1.6.6 from Arduino.cc.

However as I said in that thread there might be something different in the source code that causes the problem mentioned with GsmWebClient example and the 1.6.6 version from Arduino.cc. However I tested that this problem did not appear using the 1.5.6-r2 version, that example worked fine with the DCcduino compatible board and the GSM Shield 2, therefore something in the code changed between the versions 1.5.6-r2 and 1.6.6 from Arduindo.cc that made the example did not work properly with that DCcduino compatible board as I described.

It would be interesting if you could check the GSM libraries or other code to solve this incompatibility for those users that own compatible boards as DCcduino and GSM Shield 2 and want to use the Arduino IDE from Arduino.cc.

Many thanks for your assistance.
Best Regards

agdl, 2016-07-12:
This issue was moved to arduino-libraries/GSM#3

closed 2016-07-12 by agdl