arduino/Arduino #4072: WebServer example doesn't compile on DUE

qbahn21, 2015-11-03:
Hi.
I got an error: 'SPI_CS' was not declared in this scope... There is no defnition of 'SPI_CS' in 5100.h I've found: ETHERNET_SHIELD_SPI_CS only.

cmaglie, 2015-11-05:
@qbahn21 thanks for the report, it is now fixed in version 1.1.1.
The release should be available from library manager within an hour.

closed 2015-11-05 by cmaglie

qbahn21, 2015-11-06:
Hello.
Thanks for info. I don't see any new release right now. The latest release
is 1.0.5.

cmaglie, 2015-11-06:
uhm... that's really weird, my minimum expectation is that you have 1.1.0 installed because is the release that shows the bug. Here is what I see from the library manager:

[image]

so the release `1.1.1` has been published.
It may be that the library_index.json is not updated for some reason (file permission)? Does the IDE shows errors when you open the library manager?

qbahn21, 2015-11-06:
No, it doesn't.
I have a several Arduino versions folders (installed one and few non
admin). Has it influence? I think it has, because I see, that there is
Ethernet version 1.1.0 in 1.6.6 library folder. I think that GUI reads
config from an old (1.0.6) installed version folder
(Users...\AppData\Local...). Is there any possibility to change this
behaviour?

cmaglie, 2015-11-06:
The library manager lists the libraries found in the sketchbook folder, but I think that there is something more in your installation that is "confusing" the lib manager.
Let's try this:
* turn on verbose compile from preferences
* compile an Ehternet library example
* copy & paste here the output of the build, in particular the last part that shows all copies of Ethernet libraries available (and which one has been selected by the IDE).

qbahn21, 2015-11-06:
Ok, let's try this method :) Error log in attachment.

[image: Obraz w treści 2]

Arduino: 1.6.6 (Windows 7), Board: "Arduino Due (Programming Port)"

D:\Pobrane\arduino-1.6.6\arduino-builder -dump-prefs -logger=machine -hardware "D:\Pobrane\arduino-1.6.6\hardware" -hardware "C:\Users\jdn\AppData\Local\Arduino15\packages" -tools "D:\Pobrane\arduino-1.6.6\tools-builder" -tools "D:\Pobrane\arduino-1.6.6\hardware\tools\avr" -tools "C:\Users\jdn\AppData\Local\Arduino15\packages" -built-in-libraries "D:\Pobrane\arduino-1.6.6\libraries" -libraries "C:\Users\jdn\Documents\Arduino\libraries" -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10606 -build-path "C:\Users\JDN\AppData\Local\Temp\build1a94c3fade059878028b246e557b4116.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "D:\Pobrane\arduino-1.6.6\libraries\Ethernet\examples\WebServer\WebServer.ino"
D:\Pobrane\arduino-1.6.6\arduino-builder -compile -logger=machine -hardware "D:\Pobrane\arduino-1.6.6\hardware" -hardware "C:\Users\jdn\AppData\Local\Arduino15\packages" -tools "D:\Pobrane\arduino-1.6.6\tools-builder" -tools "D:\Pobrane\arduino-1.6.6\hardware\tools\avr" -tools "C:\Users\jdn\AppData\Local\Arduino15\packages" -built-in-libraries "D:\Pobrane\arduino-1.6.6\libraries" -libraries "C:\Users\jdn\Documents\Arduino\libraries" -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10606 -build-path "C:\Users\JDN\AppData\Local\Temp\build1a94c3fade059878028b246e557b4116.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "D:\Pobrane\arduino-1.6.6\libraries\Ethernet\examples\WebServer\WebServer.ino"
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
"C:\Users\jdn\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -w -x c++ -M -MG -MP -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\cores\arduino" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\variants\arduino_due_x" "C:\Users\JDN\AppData\Local\Temp\build1a94c3fade059878028b246e557b4116.tmp\sketch\WebServer.ino.cpp"
"C:\Users\jdn\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -w -x c++ -M -MG -MP -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\cores\arduino" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\variants\arduino_due_x" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\libraries\SPI" "-ID:\Pobrane\arduino-1.6.6\libraries\Ethernet\src" "C:\Users\JDN\AppData\Local\Temp\build1a94c3fade059878028b246e557b4116.tmp\sketch\WebServer.ino.cpp"
"C:\Users\jdn\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -w -x c++ -M -MG -MP -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\cores\arduino" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\variants\arduino_due_x" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\libraries\SPI" "-ID:\Pobrane\arduino-1.6.6\libraries\Ethernet\src" "C:\Users\JDN\AppData\Local\Temp\build1a94c3fade059878028b246e557b4116.tmp\sketch\WebServer.ino.cpp"
"C:\Users\jdn\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -w -x c++ -M -MG -MP -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\cores\arduino" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\variants\arduino_due_x" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\libraries\SPI" "-ID:\Pobrane\arduino-1.6.6\libraries\Ethernet\src" "C:\Users\JDN\AppData\Local\Temp\build1a94c3fade059878028b246e557b4116.tmp\sketch\WebServer.ino.cpp"
"C:\Users\jdn\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -w -x c++ -M -MG -MP -DF_CPU=84000000L -DARDUINO=10606 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\cores\arduino" "-IC:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\vari [... cut ...]
Using library Ethernet at version 1.1.0 in folder: D:\Pobrane\arduino-1.6.6\libraries\Ethernet 
exit status 1
Error compiling.
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

cmaglie, 2015-11-09:
@qbahn21 
sorry for the late answer..
The IDE is taking the 1.1.0 version from the install folder:

```
Using library SPI at version 1.0 in folder: C:\Users\jdn\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.5\libraries\SPI Using library Ethernet at version 1.1.0 in folder: D:\Pobrane\arduino-1.6.6\libraries\Ethernet
```

but unfortunately this won't give any clue on why the list of available library is not being updated.
May you check the file in your `%LOCALAPPDATA%\Arduino15\library_index.json` and see if it lists the Ethernet library version 1.1.1? (try to search for `1.1.1`)

Thanks!

qbahn21, 2015-11-09:
Hi.
No problem. Meanwhile I've deleted folder %LOCALAPPDATA%\Arduino15. It
helped. So problem is fixed. Thanks!
Regards.
Jakub

NitrofMtl, 2015-12-28:
I've got the same problem of updating Ethernet lib, still stuck with version 1.0.5... I've try to delete arduino15 folder but in my case doesn't fix the problem...

anyway to install it manually maybe ??