arduino/Arduino #4075: Arduino IDE 1.6.6 does not compile code with namespaces

bmitov, 2015-11-04:
I am testing this simple code with a namespace, and I am getting "sketch_nov03a:8: error: 'namespaceTest' does not name a type" error.
This works perfectly fine in any older version of the IDE . How to enable Arduino IDE 1.6.6 to compile namespaces? I need them for my project.

namespace Test
{
}

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

Chris--A, 2015-11-04:
Confirmed in hourly build 22/10/2015

SalvorinFex, 2015-11-04:
Confirmed in Win 7 and win 8 with the release version.

ffissore, 2015-11-04:
I confirm the regression. I fear you'll have to stick to 1.6.5 for now. I'll work on it and fix will be available through hourly builds asap

ffissore, 2015-11-04:
This issue was moved to arduino/arduino-builder#44

closed 2015-11-04 by ffissore

ffissore, 2015-11-04:
Fixed in latest hourly build. http://www.arduino.cc/en/Main/Software#hourly