arduino/Arduino #4076: 1.6.6 File associations are now broken with the install

SalvorinFex, 2015-11-04:
After creating a sketch for another bug, I closed the IDE and attempted to double-click a INO file.
Upon the splash screen opening and 'loading boards' the 'starting' portion of the IDE never happens.
The splash screen closes and never loads the INO file.

bmitov, 2015-11-04:
I have the same problem. This was tested on clean virtual machine. 
When installing over existing older 1.6.5 version this IDE works correct, so this happens only on clean installations.
It has been tested in both Win7 and Win8 virtual machines, and the bug happens in both.
Also trying to start the IDE from another process fails the same way in the virtual machines!

ffissore, 2015-11-04:
Can you debug this using command prompt? Type
```
cd "C:\Program Files (x86)\Arduino\" or the correct path where your IDE is installed
arduino_debug.exe full\path\to\the\sketch\you\want\to\open
```
An error message should get printed. Can you paste it here?

SalvorinFex, 2015-11-04:
Alright this is interesting, I ran the DEBUG cmd as requested and it opening the file just fine.
Next I removed the '_debug' from the command and it was also able to open the file.

This was performed on a VM that only has had 1.6.5 on it previously.
So, the issue seems to only affect the double-clicking of the file in Windows Explorer.

bmitov, 2015-11-04:
The problem is probably when the "current path" is different as is the case of doubleclicking on file:

C:\Users\mitov>"C:\Program Files (x86)\Arduino\arduino_debug.exe"
Loading configuration...
Initializing packages...
Preparing boards...
Starting...
java.lang.UnsatisfiedLinkError: C:\Program Files (x86)\Arduino\lib\AStylej.dll:
Can't find dependent libraries
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1938)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1821)
        at java.lang.Runtime.load0(Runtime.java:809)
        at java.lang.System.load(System.java:1086)
        at cc.arduino.packages.formatter.AStyleInterface.<clinit>(AStyleInterfac
e.java:42)
        at cc.arduino.packages.formatter.AStyle.<init>(AStyle.java:54)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)

        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstruct
orAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingC
onstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
        at java.lang.Class.newInstance(Class.java:442)
        at processing.app.Editor.createToolMenuItem(Editor.java:1077)
        at processing.app.Editor.addInternalTools(Editor.java:1100)
        at processing.app.Editor.buildToolsMenu(Editor.java:803)
        at processing.app.Editor.buildMenuBar(Editor.java:568)
        at processing.app.Editor.<init>(Editor.java:249)
        at processing.app.Base.handleOpen(Base.java:875)
        at processing.app.Base.handleOpen(Base.java:860)
        at processing.app.Base.handleOpen(Base.java:856)
        at processing.app.Base.handleNew(Base.java:751)
        at processing.app.Base.<init>(Base.java:455)
        at processing.app.Base.guardedMain(Base.java:219)
        at processing.app.Base.main(Base.java:135)
C:\Program Files (x86)\Arduino\lib\AStylej.dll: Can't find dependent libraries
Cannot load native library C:\Program Files (x86)\Arduino\lib\astylej.dll
The program has terminated!

bmitov, 2015-11-04:
It appears that probably there is a missing PATH setting in the environment variables during the install... so the IDE is unable to find some of its DLLs

ffissore, 2015-11-04:
It looks like your IDE is missing a "java" folder and it's using the one you installed on your computer. Can you double check if there folder `C:\Program Files (x86)\Arduino\java` exists? Java is bundled with the IDE and there is no need to install anything

bmitov, 2015-11-04:
yes, the java folder exists.
if I do 
cd "C:\Program Files (x86)\Arduino\"
arduino_debug.exe

it starts fine

if I start from another fouler as example:
C:\Users\mitov>"C:\Program Files (x86)\Arduino\arduino_debug.exe"

it fails. so there is a missing path setting in the environment variables.

to reproduce on a clean system, type from a folder different than the Arduino:
"C:\Program Files (x86)\Arduino\arduino_debug.exe"

and you will see the bug!

bmitov, 2015-11-04:
BTW: Sent you invite on G+ Hangout so we can collaborate easier to help you track the problem. If necessary, can give you remote access to the VM or anything else you may need to easily investigate the problem...

ffissore, 2015-11-05:
I've tried installing a x64 JVM on our windows box I've tried running the IDE both from its own folder and from a different folder and tried double clicking a .ino file from the sketchbook. Everything works as expected. Can you try things like disabling your antivirus for a moment? Maybe it's interfering with windows (they usually do)

bmitov, 2015-11-05:
There is no antivirus. This is clean only Windows 7 system. Absolutely nothing else. This problem has been confirmed already by multiple people on clean Windows installs in both Win7 and Win8.
Please not that if you had older Arduino installed, in the VM and then install the 1.6.6 it works well. If you install 1.6.6 on a completely clean system it is broken.
Did you test in a clean system, where you did not have any other older software, and especially no older versions of the Arduino IDE ?

bmitov, 2015-11-05:
Again, I have system ready for you to test if you want to remote access it. Please contact me on on the Hangout, and I will give you TeamViewer access to it to test.

ffissore, 2015-11-05:
Yes, I've tried with a clean win 7 and a clean win 10: both run just fine. Tomorrow I'll also try with a w10 vm

bmitov, 2015-11-05:
Can you please look into the one I have then? A lot of people have confirmed the people, so it is there. There has to be a difference between the system you use and the VMs that I have. It actually works on my main system, but I had older versions installed on it. It does not work in my clean test VMs.
Was the system you tested really clean without even Java installed on ii? It could be a Java runtime installation issue.

SalvorinFex, 2015-11-05:
@ffissore I have also tested the double clicking on clean VMs both win 7 and win 8 and it does NOT work!  The only programs on them are some chrome installs and a previous install of the IDE 1.6.5.

Also, I have tried the CMD function from other directories, i.e, not the arduino IDE one, and that does NOT work either.  When the splash screen comes up it never gets to the "Starting" display on the bottom of the splash screen, it just closes out and the process ends.

bmitov, 2015-11-05:
SalvorinFex, do you have a really clean system with nothing but windows on it? That is what I used for the test and it did not work there.

SalvorinFex, 2015-11-05:
@bmitov I am creating a brand new one, this will take some time.

bmitov, 2015-11-05:
ffissore How did you install to test? I installed Arduino IDE 1.6.6 from the installer, and then unzipped the hourly 1.6.7 build on top of it. I did not see hourly build installer.
I will try a newer hourly build to see if this has fixed somehow...

ffissore, 2015-11-06:
After using a fresh w10 on a vm, I was able to reproduce the issue. File association is working fine, the bug is elsewhere. I'll keep you posted

ffissore, 2015-11-06:
I'm still looking for a clean solution: in the meanwhile you can add the arduino IDE path to system PATH variable. This seems a well done tutorial for doing it http://www.computerhope.com/issues/ch000549.htm

ffissore, 2015-11-06:
Ok found a solution and pushed the fix. Fix will become part of 1.6.7 and will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

Thank you all for patience and continuous feedback!

closed 2015-11-06 by ffissore

SalvorinFex, 2015-11-06:
@ffissore That's good to know, Thank you for your help.

SalvorinFex, 2015-11-06:
@ffissore @bmitov I have just confirmed that the latest nightly build has fixed this, Thank you so much!

bmitov, 2015-11-06:
@ffissore I have also tested and confirmed that it is working well in the latest nightly build! Good job, thank you ! :-)

commit 53987f9d (2015-11-06): Windows: msvc*.dll are not found when loading AStyle.dll. Forcing load.
Fixes #4076