arduino/Arduino #4077: Provide a simple global search

lmihalkovic, 2015-11-04:
Finding things in a project is not easy without forward knowledge of what to look for, and even then searching is still bound by the contents of a single source file. This feature would add simple global search (scope to be defined?! ideally would support ALL reachable source files, even if the result is initially only displayed in the search result popover when the file is not editable, but ideally, any of the source files should be at least VIEWABLE)

Proposed UI is the same as eclipse/intellij
[image]

Just testing what the user experience would be (popover arrow should be right aligned?!)
[image]

Alternatively could be done via a editor-centered popup.

lmihalkovic, 2015-11-16:
The idea is to have an extensible mechanism to contribute different quick access contributors, each responding to one or more triggers (exact and partial matches). At the moment there are experimental contributors for the following
* sketch variables ( keywords "vars" "variables" - would be nice to jump to the definition when selecting )
* key bindings ( presents a partial list until more action related code is cleaned up )
* libraries ( "libs" "libraries" - at the moment lists all libraries and versions. could add 'installed libs' or 'all libs' )

[image]

[image]

Everything is running async and implementing a *computer* is dead simple

lmihalkovic, 2015-11-16:
TODO
* enhance the **vars** contributor to display the name of the library where the variable originates from (good for first time learners)
* open library reference page when selecting a variable from a library

ffissore, 2015-11-16:
Show me code, or it didn't happen http://memegenerator.net/instance/30625023

lmihalkovic, 2015-11-17:
@ffissore  Not sure what you are asking me, but I'm sure it could be done with a smile  :-)

This part is in APP
[image]

This part is in CORE (no, I did not add Token.XXXXX/RSTA lib as a dependency of CORE)
[image]

The presentation is currently in flux, and the final look I am trying to go for is something along the lines of the following two designs. I'll eventually push it to my fork when I like the result.

[image]

[image]

I appreciate that this is a creating a distraction from your tight schedule and will close this for now.

ffissore, 2015-11-17:
Nice screenshots of your code (well, actually of the file names). Mind to share the actual code? I mean the source code. Submitting a Pull Request is your best option to see your code included in Arduino https://help.github.com/articles/creating-a-pull-request/

lmihalkovic, 2015-11-17:
@ffissore  I do welcome constructive input if you feel like sharing any. I would prefer not to deal with sarcasms, but will make an effort to get used to it. This is the core input processor. I'll let you know when it's all in my fork

[image]

I use images simply because they usually reduce the need for accompanying written explanations while helping people get the true feel for how a UI works (design changes are a manageable burden if the code is cleanly structured)

ffissore, 2015-11-17:
As I've already said, I'm more than happy about your enthusiasm. However you have to understand that sharing screenshots and no actual, running code, gives everything a taste of slideware that I'm sure does not belong to your work. So please turn your contribution into something that everybody can compile and run on their computer

Testato, 2015-11-17:
It is very interesting.
Hope that the Arduino Team will pull the code from your work when it will
be available

cmaglie, 2015-11-17:
@lmihalkovic 
Looking at the amount of classes and code you wrote, this doesn't look like neither a "simple" nor a "one hour" job as you often repeat.

Moreover, I think that the cost of implementing the engines (that actually populates these GUI elements with something useful) is something to not underestimate.

lmihalkovic, 2015-11-17:
@cmaglie thank you for taking the time to comment. 

As i recall the **one hour** reference was made once (couple instances but single topic) with respect to the relooking of the ui to track Arduino Create. In support, I detailed the steps involved as well as my scope (e.g. i did not yet eliminate the bitmap file with the buttons but will replace it with as many small classes, each painting a single shape). But say I am off 3 times (my statement was accurate), we would still be looking at 3 hours for a complete relooking and convergence of the company's product line. I have an mba, hence my bias towards branding and product/corporate identity.

The actions code in the snapshot is part of a back burned task that i am doing for supporting displaying the keybindings from the quick access. I did not need to do it for quick access, but i think in the long run it will ease the maintenance burden. My other motivation for starting it was that the code for next/prev-tab and other sketch level actions is the same wether you are looking at the sketch or a library (this is all predicated on the very first change I made to the codebase that eliminated the difference between libs and sketches, making it free for the existing code to work with either). I assure you that the code itself is simple (i basically cut/past the guts of sketch.java), but i do grant you that, like in sculpture, seeing where the fracture line is in a piece of rock is not always obvious. Fortunately your codebase is quite clean, testimony to the quality of your work... which makes mine what it is.

lmihalkovic, 2015-11-17:
Strike my last paragraph... your codebase is really easy to enhance... maybe not obvious, but definitely easy because it has a good foundation.

[image]

lmihalkovic, 2015-11-19:
Added grouping of entries per libraries 

Added code to work from the current editor selection via META-3 (yes, i use eclipse on a mac). If there is no selection, then it just jumps to the field. If there is a selection, the QA tries to find a matching var/func in any of the libs. If there are any maches, then up/DOWN to select one, then ENTER to open the internal help browser (should probably add a preference to use your desktop browser instead) to the reference page for the library that contains the definition.

lmihalkovic, 2015-11-21:
Final look for now
[image]

lmihalkovic, 2015-12-04:
[LMDONE]