arduino/Arduino #4078: compile warnings from SoftwareSerial from use of always_inline

ahdavidson, 2015-11-04:
When compiling a sketch that uses SoftwareSerial with Arduino 1.6.5 and compiler warnings set to default, I get the following warnings:

/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/SoftwareSerial.cpp:375:6: warning: always_inline function might not be inlinable [-Wattributes]
 void SoftwareSerial::setRxIntMsk(bool enable)
      ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/SoftwareSerial.cpp:121:6: warning: always_inline function might not be inlinable [-Wattributes]
 void SoftwareSerial::recv()
      ^

NicoHood, 2015-11-05:
From what I remember in my always inline attempts:
* You need to mark the definition inline `inline void foo  __attribute__((__always_inline__));`
* You need NOT to mark the implementation inline `void foo { code }`

So basically I think you need to move the `inline` marker from the .cpp to the .h file. (Havent tested it=

sandeepmistry, 2015-11-05:
Closed via 12ebdfe8a09aabb4c34570a26ca80424b2e66fd2

closed 2015-11-05 by sandeepmistry