arduino/Arduino #4079: OSX 10.10.1 using Arduino 1.6.6: build message is wrong

i-make-robots, 2015-11-04:
```    Makelangelo-firmware/firmware_ams/firmware_ams.ino: In function 'void adjustSpoolDiameter(float)':
    firmware_ams:226: error: 'void adjustSpoolDiameter(float)' was declared 'extern' and later 'static' [-fpermissive]
     static void adjustSpoolDiameter(float diameter1) {
                                                    ^
    firmware_ams:226: error: previous declaration of 'void adjustSpoolDiameter(float)' [-fpermissive]
     static void adjustSpoolDiameter(float diameter1) {
          ^
    exit status 1
    'void adjustSpoolDiameter(float)' was declared 'extern' and later 'static' [-fpermissive]```

The keyword 'extern' does not appear in my code anywhere.  Instead I removed the keyword 'static' and Arduino was able to finish compiling.  This code was compiling fine on 1.6.4.

ffissore, 2015-11-05:
I've just cloned your repo https://github.com/MarginallyClever/Makelangelo-firmware and this other repo https://github.com/i-make-robots/Stewart-Platform-Crane. Then moved  `Stewart-Platform-Crane/libraries/AFMotorDrawbot/` into my sketchbook `libraries` folder. Selected Arduino/Genuino Uno and the sketch successfully compiled.
Can you add more details?

i-make-robots, 2015-11-05:
Makelangelo firmware was already fixed when I reported the error. Add the keyword "static" to any method in the firmware and try to recompile.

ffissore, 2015-11-06:
Ok, arduino-builder omits `static` when generating prototypes. Moving the issue. Anyway, please note that your sketch won't compile anymore with latest hourly because you put libraries folders inside sketch folder. Released 1.6.6 was supposed to do recursive compilation (see https://github.com/arduino/Arduino/issues/1004#issuecomment-153891462 and subsequent comments). Hourly (and 1.6.7) will attempt to compile those libraries code and will fail when compiling examples, because the library is not where it's supposed to be and the IDE can't find it

ffissore, 2015-11-06:
This issue was moved to arduino/arduino-builder#47

closed 2015-11-06 by ffissore

ffissore, 2015-11-06:
Fixed in latest hourly build http://www.arduino.cc/en/Main/Software#hourly

i-make-robots, 2015-11-06:
The instructions I give on my website tell people to move the libraries to
the correct location.  It's easier to distribute them as a single download.

Thanks,