arduino/Arduino #4082: [SAM core 1.6.5] 'max' was not declared in this scope

per1234, 2015-11-04:
Using Arduino IDE 1.6.7 2015/11/04 03:53 with SAM Boards 1.6.5 and Windows 7 64 bit

Maybe the same issue as: https://github.com/arduino/ArduinoCore-samd/commit/a546da22f9058497fcdbc9b276e21ff833f6a28e

Sketch:
```
byte x = max(1, 2);
void setup() {
}
void loop() {
}
```
Error:
```
sketch_nov04a:1: error: 'max' was not declared in this scope
 byte x = max(1, 2);
                  ^
exit status 1
'max' was not declared in this scope
```
Compiles fine with SAM Boards 1.6.4

Reported at http://forum.arduino.cc/index.php?topic=357312.msg2463840#msg2463840

odbol, 2015-11-05:
haha just found this same problem. 

Here's a bit more info, seems that there are some redefinitions loose in the USB_device.h file:

````
In file included from /Users/tyler/Library/Arduino15/packages/arduino/hardware/sam/1.6.5/system/libsam/chip.h:66:0,
                 from /Users/tyler/Library/Arduino15/packages/arduino/hardware/sam/1.6.5/cores/arduino/Arduino.h:42,
                 from /var/folders/lh/s5d4j74j4kgg0cydtkjz46780000gn/T/build2e4adf3305ada44c20a2a50ec8690852.tmp/sketch/TestMax.ino.cpp:1:
/Users/tyler/Documents/Arduino/TestMax/TestMax.ino: In function 'void setup()':
TestMax:4: error: 'max' was not declared in this scope
     int newBpm = min(max(0, 5), 6);
                              ^
/Users/tyler/Library/Arduino15/packages/arduino/hardware/sam/1.6.5/system/libsam/include/USB_device.h:364:32: note: in definition of macro 'Min'
 #define Min(a, b)           (((a) < (b)) ?  (a) : (b))
                                ^
/Users/tyler/Documents/Arduino/TestMax/TestMax.ino:4:18: note: in expansion of macro 'min'
     int newBpm = min(max(0, 5), 6);
                  ^
exit status 1
'max' was not declared in this scope 
````

For me, using `Min()` worked but `min()` did not. My guess is that the precompiler is running out of recursion? For instance, this didn't work:

````
    int newBpm = min(max(0, 5), 6);
````

But this does:

````
    int newBpm = Min(Max(0, 5), 6);
````

cmaglie, 2015-11-05:
I confirm the issue, this is a regression on the SAM core, as @per1234 said, it may be something similar to the same issue appeared on the Zero.

cmaglie, 2015-11-05:
I've pushed a fix for this bug:
https://github.com/arduino/Arduino/commit/f678ad423ac6702df4f80c8d0bf05052f5052b93

This is a major bug, we are probably going to do another release of the SAM core only to fix this one, please check if this fix is effective.

per1234, 2015-11-05:
The fix works for me with my simple test sketch and the sketch from the user who originally reported the issue. I will ask them to test also and report their findings if they don't have a GitHub account.

per1234, 2015-11-06:
The user who reported the issue on the Arduino forum says the fix resolved their problem with max() and Due: http://forum.arduino.cc/index.php?topic=357312.msg2466769#msg2466769

cmaglie, 2015-11-06:
The fix has been released.

@per1234 
Thanks for the great feedback!

closed 2015-11-06 by cmaglie