arduino/Arduino #4083: Provide a sketch navigation/outline view

lmihalkovic, 2015-11-05:
Even something seemingly simple as a single file sketch is in reality a more complex application silently pulling on multiple resources under the surface. The idea here is to be able to easily visualize these resources, without taking away any of the current beautiful simplicity. 

Well done, this feature is the doorway to many other possible extensions.

Proposed reference design:
[image]

Expanded view (*references* is inappropriate)
[image]

lmihalkovic, 2015-11-26:
Done

closed 2015-11-26 by lmihalkovic

lmihalkovic, 2015-12-09:
I am at the point where i will be able to automatically add the #include statements to the sketch source code when the user clicks on a checkbox near the name of a lib in the outline view [chk boxes to ba added soon] or automatically heckbthe right boxes when the sketch is loaded, and remove them when a #include is removed.
The point is code or clicks are fully interchangeable. And the include search path will be passed accordingly (i have my own compiler that supports that)

lmihalkovic, 2015-12-09:
Inalso have a experimental [read ugly read boxes] for moving references up and down in the outline view to decide on the link order. Default behavior is the order of the #include statement in the code (moving the refs up/down) does the same to the #includes in the source file

lmihalkovic, 2015-12-09:
Lastly, i was thinking about making the invludes section of a sketch read only with a nice box around to signify that although it is source code, the ide owns it.. But dunno if wiuld be useful or confusing

twanek, 2017-11-11:
hello!

i know this is old topic, but there is any progress / plans on implementing vertical tabs in ide?
the horizontal tab arrangement really sucks, i work with 30+ tabs, and despite i use a 24" display, i can't see all of them... always have to scroll through all the tabs to find one.

thanks!

facchinm, 2017-11-13:
Hi @twanek , would you mind trying the Beta 1.9.0? While the horizontal, fixed width tabs are still there, at least they are scrollable and do not disappear when the window is too small. Let me know if it improves your workflow or if it could be further improved, thanks!

twanek, 2017-11-13:
thanks for the reply!

sure, i can try it. should i download this beta from the arduino website?

twanek, 2017-11-13:
- the name of the tabs should have better contrast / bigger fonts, it is somewhat hard to see the text

[image]

- the "new tab" button is floating over the tabs, and it is too small. it would be better to place NEAR the serial monitor button, not UNDER. anyway, there is plenty of unused space in that line with the serial monitor button.

- if a tab which is not visible in the current view it is scrolled to, or selected from the drop down list, it is showed in the very right of the screen. this is not too practical + it is partially covered by the "new tab" button

- the "current tab" highlighting and the scroll function is really cool. but in the long run, vertical tabs are needed for big projects

- as a quick - but usable - workaround for vertical tabs, you could just add a "pin" option to the "new tab" drop down list: it should make the drop down list persistent, floating over the code editor area. it would be a real treat!

[image]

- what about natively integrating this nice dark theme?

[image]

hope this feedback is useful, thanks for the work! if i can help with anything else, just send me a message.