arduino/Arduino #4084: Give some coolness factor to the ArduinoIDE

lmihalkovic, 2015-11-05:
Aside from the bang-on-feature-set and quality of the board, IMHO the utterly simple approach of the supplied IDE is what made the Arduino a hit... instead of the dark-art that embedded programming has been for years, it suddenly became possible for any teenager to download a single zip and reliably get going in minutes. This is a awesome achievement!!! As time has passed, the IDE as retained the simplicity that made its success while adding capabilities. 

IMHO the platform is coming to a cross-roads. The skill-set of programmers keep growing, more rapidly than before I might add, and the current IDE is approaching (some might say it passed it) a point where its minimum feature-set will not longer be enough.

This issue entry is for the purpose of referencing under a single umbrella other individual issues meant to inject a number of capabilities that meant to give a dose of coolness factor to the current IDE.

* code completion
* sketch navigation/outline view  #4083 (touching #4050)
* sketch debugger  #4010 
* global search  #4077
* user defined scripts (think file validation, data visualization, ...)
* code templates
* wizards (sketch, library)
* dedicated class-room support features

More practical
* library editing (intersecting with #4065 #4065)
* better multi-tab handling  #3061

ffissore, 2015-11-05:
Thank you @lmihalkovic I'm gonna close this one as it's not really an issue but rather a collection of proposals. Please join the developers mailing list where you can share your priorities with other arduino hackers and receive their feedback and points of view

ffissore, 2015-11-05:
Forgot to add the url https://groups.google.com/a/arduino.cc/forum/#!forum/developers

closed 2015-11-05 by ffissore

lmihalkovic, 2015-11-05:
Yes, thanks for the pointer, I will join.

NicoHood, 2015-11-05:
You always show screenshots of those features.
If you implemented them already, why not open a PR or at least upload the fork changes so people can recompile the IDE and try it out? Or am I missing something?

lmihalkovic, 2015-11-05:
I recently got an arduino, and in the process of discovering it, I started running into pain points with the IDE. Considering the maturity of the IDE I figured I couldn't be the first one... Except for the debugger, these are small tasks requiring remarkably little work with the editor the way it is today. Having a bit of time on my hands, I decided to write code partially to test the ideas, partially to make myself more productive, partially to make others more productive (this is very subjective as it assumes that my solutions are general enough that they make sense to others).

My time for this effort is very limited to a few hours per day, and I will soon have to return to other things altogether. I work from a series of branches that I rebase against the official code. Some of the features would be moving faster with a bit of help (mostly simply question answers) from people who know the environment better than I do (for e.g. I still don't know if there is a keywords.txt for the code platform code, and only started working on .S completion tonight when I realized ASM code also works in the editor). So far I have answered more questions than even the simplest of mine have received answers.

I have not taken the time to push my tree to my GH fork, and I am not creating PR because either takes time that this week I'd rather spend making progress. On the topic of PRs, I ran into #3484 today. It seems the work spans over the entire year 2015, with some creation date going even further back in time. Other PRs seem to have lingered for variable durations.. I did not see an urgency to me pushing code I started to write last week.

NicoHood, 2015-11-06:
Thanks for the answer. I really think you should share the code if you want them to fix it. Even if the patch is not ideal, they might have a simple idea to fix it another way. This helps more than just asking for those features. Also the time of the devs is limited, but trust me, they are doing the best they can. I am hoping that you features will be included to the IDE :)