arduino/Arduino #4085: Support for editing library code

lmihalkovic, 2015-11-05:
Libraries are a great way to enhance the platform popularity. There is not good technical reason why the IDE cannot be altered to support editing them.

Proposed reference design
[image]

(ps: this is actually a working implementation)

matthijskooijman, 2015-11-05:
Cool! I might have missed it, but did you have a PR open with this code? Or do you perhaps just have it pushed to github somewhere so I can take a peek?

lmihalkovic, 2015-11-05:
As I wrote somewhere else, I was having some technical challenges (read: I was twisting myself in the ropes) with altering arduino-builder to support that feature and the customizable link priority scheme I also described, so I had to back-track and implement my own compiler. The redeeming factor is that I made it a config param to switch between all 3 compilers.

matthijskooijman, 2015-11-05:
What changes do you need in arduino-builder for this? To allow modified-but-not-saved library files to be included in the build?

lmihalkovic, 2015-11-05:
At the moment the IDE uses CTL-L to go to a given line of code. In eclipse the default is to use CTRL-G (goto). In the library editor I was thinking about (read: am currently) using CTRL-L for opening the library's folder in the OS navigator (Finder/Explorer/... on Linux).

[image]

Does anyone have a better suggestion for OpenLib, or might it be possible to switch CTRL-L to CTRL-G instead?!

lmihalkovic, 2015-11-26:
Done

closed 2015-11-26 by lmihalkovic