arduino/Arduino #4088: Invalid place of generated forwards

oscaracena, 2015-11-05:
Hi, 

There is a problem when Arduino **generates** the .cpp from a sketch. It usually takes all the function definitions and place a **forward declaration** of them just before the first no-preprocesor directive. This is problematic when the forward declaration uses a **struct** (for example), because it will place the FD before the struct declaration. The following code would not compile:

```cpp
typedef struct {  
} A;

void fn(A* a) {}
void setup() {}
void loop() {}
```

If you include the forward declaration, it will fix the problem, BUT it must be written exactly as its definition. I.E, this will compile:

```cpp
typedef struct {  
} A;

void fn(A* a);
void fn(A* a) {}
void setup() {}
void loop() {}
```

while this NOT:

```cpp
typedef struct {  
} A;

void 
fn(A* a);

void fn(A* a) {}
void setup() {}
void loop() {}
```

I'm not sure if this is a documented problem. I've searched but not found anything.
Thanks!

ffissore, 2015-11-05:
Which version of the IDE are you using?

oscaracena, 2015-11-05:
Ooops, sorry. I'm using 1.6.5, because I use a package that is not ported to 1.6.6 yet. I should have checked against 1.6.6 before posting the issue. I see that this problem is fixed.

Sorry again and thanks for pointing out!
Cheers.

ffissore, 2015-11-05:
Glad to see a already fixed issue ;) Thank you

closed 2015-11-05 by ffissore