arduino/Arduino #4089: Provide code completion inside editor

lmihalkovic, 2015-11-05:
The IDE currently does not provide any form of CTRL-SPACE code completion. The feature is sorely missing, especially for newbies like me. The feature should cover the following

* basic completion based on KWxxx and LITyyy available in the sketch libraries
* specific completion for different file types (keywords.txt, dependencies.txt, library.properties)
* a clean framework for future file types, ....

lmihalkovic, 2015-11-05:
Initial implementation working for keywords.properties

[image]

[image]

[image]

At the moment I am not sure what LITERALXXX are for so the name strings are all wrong. The implementation is also missing the typical explanation window that opens when a keyword has been selected.

lmihalkovic, 2015-11-05:
Current completion for library's library.properties

[image]

ffissore, 2015-11-05:
Duplicate of #849 and #3484
@lmihalkovic it seems like you have running code already. Mind to show it? The smaller the PR, the easier it gets merged

lmihalkovic, 2015-11-05:
I seemed to have missed an episode... where can I find a keywords.txt for the code platform code?

closed 2015-11-05 by ffissore

lmihalkovic, 2015-11-05:
@ffissore I read #3484... quite the PR. IMHO integrating it into the core code as it is basicaly equates giving up on a lot things that are currently easy to do and signing up for a lot of maintenance headaches. The fact that I think it is does not make it so, but the fact that this PR requires so many changes to editor and textarea should itself be a smell if poor API design. 

I just find truly unfortunate that a lot of good existing design might be sacrificed.

lmihalkovic, 2016-04-27:
Interestingly enough, the weird highlighting in the screenshots was caused by some of the issues I was alluding to in my last comment (code in the _wrong_ place).