arduino/Arduino #4090: Serial plotter blocks monitor

tigoe, 2015-11-05:
The serial plotter is great, but it blocks the serial monitor, and there's no way to copy/paste values from it. 

As an example:
1. Open File -> Examples -> Basics -> AnalogReadSerial, and upload to a board
2. Open Plotter
3. use whatever analog sensor is attached to A1.

The plot will move too fast to see the changes, and you can't scroll back and look at details. That's a useful feature of a plotter. Perhaps it could have an autoscroll button like the serial monitor does.

Also, there's no way to capture the serial data when the plotter's open. This is a feature that's been very useful from the serial monitor for years. 

As a fix, I suggest 
(a) adding a button that has the same functionality as autoscroll in the serial monitor
(b) adding the ability to copy the serial output from the plotter, same as in the serial monitor

azjata, 2015-11-05:
serial plotter is something you can write in 30 minutes. stop using arduino ide as it's a toy for a 5 year old written in processing

henningpohl, 2015-11-11:
This is due to the serial connection object being a part of the serial monitor instance. This pattern was kept in place for the serial plotter. However, only one connection can be opened per port. To have both the plotter and the monitor open, the serial connection would need to be managed elsewhere. This could be a SerialConnection singleton, or some other mechanism. Both plotter and monitor could then subscribe to serial events. As that is a bigger refactoring, I didn't touch this when writing the serial plotter.

tigoe, 2015-11-11:
Fair enough. It just makes the plotter less useful than it may seem at first appearance. given that, I would at least implement the autoscroll stop/start.

madsdyd, 2020-04-07:
@per1234 Mayybe this should be labelled "SerialPlotter" too?

madsdyd, 2020-04-07:
Any idea how hard it would be to refactor SerielPlotter to be a text monitor as SerielMonitor? Purpose would be to have a text pane (hideable) that would show the raw serial output.

madsdyd, 2020-04-07:
This is really 2 issues in one:

* Ability to pause the serial plotter: This is in #4860 (and have a workaround, actually)
* Ability to capture the serial input when the plotter is running and/or have access to the serial monitor at the same time as serial plotter is open.

To make this clearer, I suggest that this bug is solely about bullet to, and ways to work around/solve this problem.

cousteaulecommandant, 2020-04-10:
> Any idea how hard it would be to refactor SerielPlotter to be a text monitor as SerielMonitor? Purpose would be to have a text pane (hideable) that would show the raw serial output.

Sounds nice, but if this is being done it makes no sense to keep the old Serial Monitor.  Just have a single tool that combines Serial Monitor and Serial Plotter, with tabs or buttons to switch between "text view" and "plot view".  "Serial Monitor" menu entry would just open this tool in text view, and "Serial Plotter" in plot view.

per1234, 2022-11-18:
Resolved in Arduino IDE 2.x