arduino/Arduino #4091: Board Manager Oddities

BrentWilkins, 2015-11-05:
A few people, including myself, have notice this error string when trying to compile for a platform added via the boards manager: "Board {0} (platform {1}, package {2}) is unknown" - arduino-core/src/cc/arduino/Compiler.java:64

Not everyone I've asked can reproduce this error state. For me it eventually went away w/o me fixing anything. The only thing I recall doing besides comparing strings to try and debug my board and platform files was to install another board using a separate JSON file in the board manager. This took a few open and closes of the board manager before the new item showed up in the list available for install.

Sorry this isn't reliably reproducible, but it may provide some insight to whatever the real issue is. This issue has been seen in Windows 8.1 and OS X (10.10.5).

ffissore, 2015-11-06:
Thank you @BrentWilkins, looking into that

ffissore, 2015-11-18:
I could find why or how that error message could have been generated. Do you have any newer hint?

ffissore, 2015-11-18:
"I could **not**". Sorry for the typo

BrentWilkins, 2015-12-04:
I haven't messed with the Board Manager in a while, so sorry, no.

agdl, 2016-07-25:
I’m closing this for now, until we have more info on how to reproduce it

closed 2016-07-25 by agdl