arduino/Arduino #4093: SparkFun_MAX31855K_Thermocouple_Breakout_Arduino_Library

BrentWilkins, 2015-11-05:
Will you please add the following library to the library manager?

https://github.com/sparkfun/SparkFun_MAX31855K_Thermocouple_Breakout_Arduino_Library

ffissore, 2015-11-06:
Added. Thanks!

closed 2015-11-06 by ffissore

BrentWilkins, 2015-12-04:
It looks like this was possibly the first case of not adding new libraries to current (and old?) versions of the IDE. From my perspective this lowers the value to the user. They won't be able to use the library manager for newer libraries & will have to manually install them for several months. Is this a permanent change?

BrentWilkins, 2015-12-04:
This looks like it might be an example of issue 4091. We had several machines, two here, and a customer's that wouldn't show this library in the library manager. I've now been able to find the library in the manager on two other machines now. Windows vs Mac doesn't seem to make a difference.

ffissore, 2015-12-07:
Hi @BrentWilkins I can find and install this lib without issues. I've also compiled its example for good measure. I'm not sure what's the problem here.