arduino/Arduino #4094: Due: SerialUSB port busy after disconnecting and trying to reconnect

odbol, 2015-11-05:
This is a regression in 1.6.6. 1.5.5 works correctly. 

On Arduino Due, upload any sketch:

````
void setup() {
  // put your setup code here, to run once:
SerialUSB.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  SerialUSB.println("hello");
  delay(100);
}
````

1. Upload it to the board using Native USB port.
2. Run Serial monitor.
3. It should connect and display the output correctly.
4. Close Serial Monitor.
5. Open Serial Monitor again.

Expected: it connects correctly.

Actual: get a `Error opening serial port '/dev/cu.usbmodem1d111'. (Port busy)` error. This happens in other apps too, like "Serial Tools" for OS X. It seems that when you close the connection on the PC side, the Arduino is not properly cleaning up and freeing the connection.

Tested on Arduino Due, OS X 10.10.5, with Arduino 1.6.6 (Due core 1.6.5).

odbol, 2015-11-06:
Oops, looks like this also works in Arduino 1.6.2, so it must have been a change after that. (Specifically, git revision 72bf97709c148abba34dd2455224e7af36f0294e)

sandeepmistry, 2015-11-13:
Confirmed, I've narrowed it down to the following commits: 2acda414aa06e2d5bbdc5394bb99836e8d6c4ec0 or a261e6cf7c6bd0c9d3442685700e79b20ee154bd

sandeepmistry, 2015-11-13:
If I go back to b2381506a253cae5d3ad8b5dfd9a434c7386cfcf and [disable ```HID_ENABLED```](https://github.com/arduino/Arduino/blob/b2381506a253cae5d3ad8b5dfd9a434c7386cfcf/hardware/arduino/sam/cores/arduino/USB/USBDesc.h#L21) I see the same issue as well.

odbol, 2015-11-17:
May be related: https://forum.pjrc.com/threads/31043-Teensyduino-1-26-Beta-1-Available

sandeepmistry, 2015-11-17:
I've tracked it down to OS X sending a ```CDC_SEND_BREAK``` on close, and the SAM core not handling it correctly.

The following patch corrects the ```CDC_SEND_BREAK``` handling: https://github.com/arduino/Arduino/pull/4171

PR #4171 (2016-01-18): Port Serial_::readBreak() API to SAM core
Port of the AVR ```Serial_::readBreak()``` API (https://github.com/arduino/Arduino/commit/b58f2394005cd073d582b8f90f8ba222aa95b0ee) to the SAM core.

Resolves #4094, as SAM core was not correctly handling receiving a ```CDC_SEND_BREAK``` and sending a stall, causing OS X to report the serial port as "busy". (OS X sends the break when closing the serial port).

cc/ @matthijskooijman
closed 2016-01-18 by cmaglie