arduino/Arduino #4098: Board manager preferences URL doesn't take consider file://

aethaniel, 2015-11-06:
While doing the setup for a given package, having the opportunity of using local URL file:// would be highly useful instead of pushing the (maybe wrong) files to a server.

Is it possible to obtain this?

ffissore, 2015-11-11:
Yes we can add file:// support. In the meantime, you may want to rename your `package_myboard_index.json` file to `test_package_myboard_index.json`. The IDE will load that file as it was downloaded. Maintainers name protection still applies: because your test json file is not signed, you won't be able to use `arduino` or `Intel` platform names

aethaniel, 2015-11-19:
Hi @ffissore, I have put test_package_....json file in /Arduino15 and it is indeed parsed.
Everything worked well after copying the corresponding package archive in 'staging' subfolder.

Thanks for the dev tip!

ffissore, 2015-11-19:
Next hourly build will support file:// protocol as well. http://www.arduino.cc/en/Main/Software#hourly Thank you for the proposal

commit 96091879 (2015-11-19): Added support to file:// protocol. Fixes #4098
closed 2015-11-19 by ffissore