arduino/Arduino #4100: Point Release note blog post to HID master

NicoHood, 2015-11-06:
I merged the changes now to the master branch. A library manager release will follow.
Please relink to this link:
https://github.com/NicoHood/HID

Talking about this blog post:
https://blog.arduino.cc/2015/11/03/arduino-ide-1-6-6-released-and-available-for-download/

Thx for all the cooperation with the USB Core, it turns out to be very useful and simple for a (normally) complex funktion like an USB Core. I am really happy about the progress we did.

**Please also add my library to the library manager.**
There are currently releases which wont work (since it was no library before) but I will tag another release soon.

facchinm, 2015-11-09:
Hi Nico, blog post updated, assigning to @ffissore for Library Manager inclusion!

cmaglie, 2015-11-09:
I've added the library, but the tag contains a `.development` file, you should remove the file and re-tag the release. Once done the library should be published within an hour.

closed 2015-11-09 by cmaglie

NicoHood, 2015-11-09:
whoops. I will release a new version then, does this work too? I dont want to mess with the git again.

cmaglie, 2015-11-09:
yup that works too