arduino/Arduino #4101: Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern

SalvorinFex, 2015-11-06:
I am getting this on a fresh Wn 7 install after adding the SAM boards via the board manager.

This has happened twice today on different systems with IDE 1.6.6 .

_Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core._

NicoHood, 2015-11-06:
As stated in the error `Consider upgrading this core.`

SalvorinFex, 2015-11-06:
The Boards for the SAM are already at 1.6.5, which is as high as it goes.

cmaglie, 2015-11-06:
Hey, just released SAM 1.6.6 with the fix (plus another major bugfix), you anticipated me by a minute!

closed 2015-11-06 by cmaglie

SalvorinFex, 2015-11-06:
So, how do we get that into our IDEs?

cmaglie, 2015-11-06:
Through "board manager", if the update button does not appear try to close and reopen the IDE.

SalvorinFex, 2015-11-06:
Okay, the first few times I tried that it didn't work, but now it is working...Thank you!