arduino/Arduino #4103: Sketchbook location on Portable Edition

Testato, 2015-11-07:
you know that is it possible create an official "Portable Arduino IDE Edition" simply by create a directory named "portable" inside the Arduino directory.

But in this case when you try to locate your sketch location we have 3 different behavior on the 3 supported operating systems.

Go under: File > Preferences > Sketchbook location, and whit the Browse button try to locate your Sketch folder, this is the results:

- On Linux (Xubuntu 15.10)
. you can choose every directory you need, everywhere in the file systems

- On Windows (Win7)
. you can choose every directory BUT only on the same partition where is the Arduino IDE (if you have it on C:\ is it not possible select a sketchbook dir present on D:\)

- On Mac (OS X 10.11.1)
. you can choose ONLY a dir inside the "portable" folder, if you try to change the selection you receive error like explained in this topic by gpb01
http://forum.arduino.cc/index.php?topic=153592.msg2467987#msg2467987

ffissore, 2015-11-19:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-19 by ffissore

Testato, 2015-11-19:
Fixed on Windows and Mac ?

ffissore, 2015-11-20:
Everywhere

commit 1862827c (2015-11-19): Portable folder can now be set to any arbitrary path, not just subfolders
of "portable". Fixes #4103