arduino/Arduino #4109: possible preprocessor error when checking for included files

soundanalogous, 2015-11-08:
In Arduino 1.6.5, this worked (and still works fine):

```
#if !defined ethernet_h && !defined _YUN_CLIENT_H_
#error "you must uncomment the includes for your board configuration. See OPTIONS A and B in the NETWORK CONFIGURATION SECTION"
#endif
```

[Full file for greater context](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataEthernet/StandardFirmataEthernet.ino#L124-L126).

However since updating to Arduino 1.6.6 the error is always thrown. It seems to be that `ethernet_h` is no longer found even though Ethernet.h is included. For now I'm just commenting out the above lines so the code will at least compile. The Ethernet library is still included and works as expected. I suspect this could be a possibly issue in arduino-builder. It could also just be sloppy coding on my part that is picked up by stricter checking in the updated arduino-builder.

matthijskooijman, 2015-11-08:
I think this is essentially the same issue as https://github.com/arduino/arduino-builder/issues/33 (though that talks about a specific `USBCON` define, but the issue is really more generic). I'll add a comment to that issue soon to highlight the generic issue.

ffissore, 2015-11-11:
Also https://github.com/arduino/arduino-builder/issues/51 is related. These are already fixed or getting fixed. Please keep an eye on https://github.com/arduino/arduino-builder/issues/33

closed 2015-11-11 by ffissore