arduino/Arduino #4110: installing IDE 1.6.6 fails to  find avrdude and  subsequent roll back to 1.6.5

wobbler, 2015-11-08:
my machine (Windows 10 ) that previously ran 1.6.5 with no problems and was then borked by 1.6.6 , I did a revo un installer uninstall , cleaned the registry entries and left over files , then CCcleanered the machine Un-installed Java and removed al references to it too. I then reinstalled with latest release of 1.6.6 and installed all the flagged board descriptors. I loaded blink and compiled and attempted to up load to an uno the error messages are below. Reinstalled 1.6.5 remains borked too

Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
java.io.IOException: Cannot run program "{path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
   at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
   at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
   at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:113)
   at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:157)
   at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
   at processing.app.Sketch.upload(Sketch.java:1178)
   at processing.app.Sketch.exportApplet(Sketch.java:1152)
   at processing.app.Sketch.exportApplet(Sketch.java:1124)
   at processing.app.Editor$DefaultExportHandler.run(Editor.java:2452)
   at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
   at java.lang.ProcessImpl.create(Native Method)
   at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
   at java.lang.ProcessImpl.start(ProcessImpl.java:137)
   at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
   ... 9 more
 any advice would be welcome

dcellucci, 2015-11-20:
Can confirm that this is also happening on OSX 10.10.1: 

java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": error=2, No such file or directory

wobbler, 2015-11-20:
A second reinstall of 1.6.5 is now functioning. awaiting resolution of 1.6.6.

wobbler, 2015-11-20:
sorry finger slipped didn't mean to close it.

ffissore, 2015-11-20:
@wobbler @dcellucci can you please open boards manager and verify you're using IDE 1.6.6 with latest avr core 1.6.9 ?

ffissore, 2015-11-20:
Turns out this is an actual bug and we are solving it. Follow ups in #4193

ffissore, 2015-11-20:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-20 by ffissore

wobbler, 2015-11-22:
Thank you , Issue resolved by further step following the later nightly build of removing old libraries folder form the sketchbook library , updating from the new IDE tools> board> boards manager and putting the libraries folder back in the sketchbook. folder.

professorjmk, 2015-12-19:
Am running WIN 7 SP 1.  I have installed Arduino 1.6.6 and started having this same problem: "Cannot run program "{path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)..."
On 18 DEC I installed Arduino 1.6.7 and I still have this same problem.
I checked Boards Manager and I am running version 1.6.9.
What should I do to fix this and get back programming my Arduino?

wobbler, 2015-12-19:
fixed by installing a nightly build. in verbose mode still some warnings but code compiles again. 
HTH

professorjmk, 2015-12-19:
Code does compile; just will not upload to Arduino

wobbler, 2015-12-19:
The nightly build fixed the problem with up loading for me as well as problems with compiling.

professorjmk, 2015-12-19:
Hello again.

I downloaded the latest build, extracted the files, and copied the
folders/files into the Arduino (x86) program folder.  I then ran the
Arduino IDE with my current application software and tried to do compile &
upload.

It says it compiled but stayed stuck on the "uploading", showing the
following error messages:

Board Arduino-master:arduino:bt328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_BT328
Board Arduino-master:arduino:pro doesn't define a 'build.board' preference.
Auto-set to: ARDUINO_PRO
Board Arduino-master:arduino:fio doesn't define a 'build.board' preference.
Auto-set to: ARDUINO_FIO
Board Arduino-master:arduino:leonardo doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_LEONARDO
Board Arduino-master:arduino:LilyPadUSB doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_LILYPADUSB
Board Arduino-master:arduino:atmega168 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ATMEGA168
Board Arduino-master:arduino:uno doesn't define a 'build.board' preference.
Auto-set to: ARDUINO_UNO
Board Arduino-master:arduino:robotControl doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ROBOTCONTROL
Board Arduino-master:arduino:atmega328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ATMEGA328
Board Arduino-master:arduino:mini328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_MINI328
Board Arduino-master:arduino:pro328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_PRO328
Board Arduino-master:arduino:atmega8 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ATMEGA8
Board Arduino-master:arduino:mini doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_MINI
Board Arduino-master:arduino:mega2560 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_MEGA2560
Board Arduino-master:arduino:robotMotor doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ROBOTMOTOR
Board Arduino-master:arduino:nano328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_NANO328
Board Arduino-master:arduino:ethernet doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ETHERNET
Board Arduino-master:arduino:esplora doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_ESPLORA
Board Arduino-master:arduino:lilypad doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_LILYPAD
Board Arduino-master:arduino:pro5v328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_PRO5V328
Board Arduino-master:arduino:bt doesn't define a 'build.board' preference.
Auto-set to: ARDUINO_BT
Board Arduino-master:arduino:pro5v doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_PRO5V
Board Arduino-master:arduino:micro doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_MICRO
Board Arduino-master:arduino:nano doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_NANO
Board Arduino-master:arduino:mega doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_MEGA
Board Arduino-master:arduino:diecimila doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_DIECIMILA
Board Arduino-master:arduino:lilypad328 doesn't define a 'build.board'
preference. Auto-set to: ARDUINO_LILYPAD328
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated
compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically
converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider
upgrading this core.
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated
tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude,
automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude.
Consider upgrading this core.
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated
tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf,
automatically converted to
tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this
core.
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags}
{compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}",
automatically converted to
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags}
{compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider
upgrading this core.

Sketch uses 2,886 bytes (8%) of program storage space. Maximum is 32,256
bytes.
Global variables use 192 bytes (9%) of dynamic memory, leaving 1,856 bytes
for local variables. Maximum is 2,048 bytes.
java.io.IOException: Cannot run program "{path}/bin/avrdude": CreateProcess
error=2, The system cannot find the file specified
    at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
    at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
    at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:123)
    at
cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:167)
    at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
    at processing.app.Sketch.upload(Sketch.java:1178)
    at processing.app.Sketch.exportApplet(Sketch.java:1152)
    at processing.app.Sketch.exportApplet(Sketch.java:1124)
    at processing.app.Editor$DefaultExportHandler.run(Editor.java:2430)
    at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot
find the file specified
    at java.lang.ProcessImpl.create(Native Method)
    at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
    at java.lang.ProcessImpl.start(ProcessImpl.java:137)
    at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
    ... 9 more

Please let me know what to do to fix this.  Do I need to install JAVA?  If
so, what version/flavor of JAVA?

Thank you for your assistance.

MatanRobo, 2015-12-21:
Hi,

I downloaded the 1.6.7 version yesterday(20.12), and tried to upload a sketch to arduino pro-micro. I added SparkFun folder to the Hardware folder. Get the same error:

Cannot run program "{path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified

Any idea what is the problem and how solve it?

cmaglie, 2015-12-21:
> I downloaded the latest build, extracted the files, and copied the folders/files into the Arduino (x86) program folder. I then ran the Arduino IDE with my current application software and tried to do compile & upload.

@professorjmk 
where do you have donwloaded the latest build?
did you get the hourly build from here (in the Hourly builds section)?
https://www.arduino.cc/en/Main/Software

I'm asking because the warning "Board **Arduino-master**:arduino:bt328 doesn't define a 'build.board'" makes me think that you have downloaded some snapshots from github (using the download button) that is not the correct way.

@MatanRobo 
You don't need to download and add to the hardware folder manually, sparkfun provides a Boards Manager url for their boards, please have a look here:

https://learn.sparkfun.com/pages/CustomBoardsArduino

(you probably want to remove the manually added hardware folder before trying this)

professorjmk, 2015-12-21:
I downloaded the "hourly build" ZIP folder and used the decompressed
folders and files to overlay the files in Arduino X86 program files folder.

Donlow, 2015-12-27:
So if anyone else have troubles with Sparkfun missing avrdude replace "auto corrected" entries with such from https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt

Pitel, 2016-03-09:
Just had the same problem. Pretty much clean (just starting with Arduino) install of Ubuntu 15.10 64bit, downloaded hourly build, tried to upload the Blink sketch, sketch compiled, but upload failed:

```
{runtime.tools.avrdude.path}/bin/avrdude -C{runtime.tools.avrdude.path}/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -Uflash:w:/tmp/build1f30949caa7c2df3acf4a5ab1a16c918.tmp/Blink.ino.hex:i 
java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": error=2, Adresář nebo soubor neexistuje
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:123)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:175)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1186)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2381)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, Adresář nebo soubor neexistuje
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
```

Guardado, 2016-03-30:
Same problem with 1.6.7 and 1.6.8

I have confirmed this in Windows, linux. 

I'm using a [Red Board](https://www.sparkfun.com/products/11197) from sparkfun which uses avrdude to upload programs to the board. 

`Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Bootloader file specified but missing: /home/garza/Arduino/hardware/sparkfun/avr/bootloaders/atmega/ATmegaBOOT_168_atmega128rfa1.hex

Sketch uses 712 bytes (0%) of program storage space. Maximum is 126,976 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 16,375 bytes for local variables. Maximum is 16,384 bytes.
java.io.IOException: Cannot run program "{path}/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:123)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:167)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2430)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
`

Guardado, 2016-03-30:
So sparkfun had some old platform.txt and board.txt files which didn't had some new definitions needed by the arduino IDE versions 1.6.7, 1.6.8, ...
I modified them and now everything is working good. 

Here are my new files, maybe it can help someone out there....

board.txt
https://gist.github.com/Guardado/11871c889dbd9f0d633a668a51cbc335

platform.txt
https://gist.github.com/Guardado/e344df4c7b88bdb8fc090b0d5c6e0bfe

Thanks!

shwetadutta100, 2016-07-14:
Switched out the provided board.txt and platform.txt and it worked immediately- thanks for the post!

freemanjiang, 2016-07-29:
Arduino 1.6.10 has the problem. How to fix it ?

java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": CreateProcess error=2, 系统找不到指定的文件。
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, 系统找不到指定的文件。
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
An error occurred while uploading the sketch

JahrudZ, 2016-08-04:
^ can confirm 1.6.10 has it on OSX

matthijskooijman, 2016-08-04:
I'm not entirely sure, but this problem might have been fixed with #5199 (which was merged yesterday). Could you try the hourly build to see if it indeed solves your issue?

lazerwalker, 2016-11-03:
I was having the same problem (with Arduino 1.6.10 and macOS 10.12). Upgrading to 1.6.12 solved it.