arduino/Arduino #4112: [Library Manager] Please add HardwareSerialRS485

MichaelJonker, 2015-11-08:
Hello, could you please add the library HardwareSerialRS485:

https://github.com/MichaelJonker/HardwareSerialRS485

This library offers half-duplex, concurrent multi-drop (i.e. multi-master, multi-slave) RS485 communication, with message addressing and filtering, as well as collision detection and collision avoidance.

Thanks,
Michael Jonker

cmaglie, 2015-11-09:
Done, thanks!

closed 2015-11-09 by cmaglie

MichaelJonker, 2015-11-09:
Dear Christian,

thanks for adding my library to the library manager. I checked the library installation and I see that all is ok with the code. Unfortunately the URL in the library properties file was incorrect and hence there is no easy access to the documentation.
I have corrected this and released version 2.0.1 using releases under github about six hours ago.

Now I have some doubts whether I now should push the new release again as an issue or whether the librarian will eventually pick it up.
The documentation is not so clear, what I understood is that a librarian jobs runs every hour to look for updates and makes those available.

Please let me know if I should do anything special.

Thanks

Michael Jonker