arduino/Arduino #4114: Arduino AVR Boards 1.6.9 incompatible with IDE<1.6.6

per1234, 2015-11-09:
Arduino AVR Boards 1.6.9 contains the following recipe:
```
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"
```
Which is incompatible with any IDE previous to 1.6.6. After updating to 1.6.9 any compile using these boards causes the error:
```
avr-gcc: error: C:\Users\per\AppData\Local\Temp\build8524536844090177810.tmp/core.a: No such file or directory
```
This is a perfect example of why https://github.com/arduino/Arduino/issues/3932 is needed.

Chris--A, 2015-11-09:
Maybe this is related: https://github.com/arduino/Arduino/issues/4110

When the new core downloads, it ends up in the AppData\Local\Arduino15 folder and must override the copy packaged in the install folder.

per1234, 2015-11-09:
>When the new core downloads, it ends up in the AppData\Local\Arduino15 folder

Using Boards Manager with <=1.6.5 it installs to AppData\\**Roaming**\\Arduino15

per1234, 2015-11-09:
But that's Windows only. With OS X or linux I believe 1.6.6 and 1.6.5 use the same arduino15 folder so if you install 1.6.6, get the boards update notification, update Arduino AVR Boards to 1.6.9, and then decide you want to go back to 1.6.5 it seems like installing 1.6.6 broke 1.6.5 even though it was really updating Arduino AVR Boards which could have happened in either version of the IDE.

per1234, 2015-11-15:
Arduino SAM Boards 1.6.6 is also incompatible with Arduino IDEs previous to 1.6.6.

AndKe, 2016-03-21:
still not fixed in boards v1.6.10

sandeepmistry, 2016-07-18:
Now that #4737 is merged newer core releases should not have this issue, there's not much we can do about AVR core 1.6.9 and SAM 1.6.6 so closing.

closed 2016-07-18 by sandeepmistry