arduino/Arduino #4115: Arduino Projects Book: Windows Installation needs update

davidcornell123, 2015-11-09:
Page 16 of the projects book provides information for Windows Installation.  Since the printing date is 2013, it cannot include facts for Windows 10, but should be updated for a newer printing..

This advises users that Arduino software installation completed with no problems under Windows 10.  When the Arduino Uno board was plugged into the USB port, the driver installed automatically, assigning COM 4 to the device.  Communication following pp. 18-19 worked perfectly.  So installation under Windows 10 is simpler and more automatic than with earlier versions of the Microsoft OS.

SalvorinFex, 2015-11-10:
This GitHub repository is for the IDE not the book, your best bet would be to contact the publisher.

davidcornell123, 2015-11-10:
Thanks Kurt,  I've closed the publishing issue on GitHub.
Dave Cornell

closed 2015-11-10 by davidcornell123

SalvorinFex, 2015-11-10:
You're welcome and BTW, your reply shows your address and number number.
Just an FYI

davidcornell123, 2015-11-10:
Yes, I customarily send out signature block with that info.  Do you 
think I should be more guarded about personal contact info?

Dave

SalvorinFex, 2015-11-10:
I am but that's up to you, especially the phone number.  
I have our phones on the "do not call lists" and we still get robocalls and other junk calls.