arduino/Arduino #4118: preprocessor issue when "{" is present

qbahn21, 2015-11-09:
Arduino 1.6.6 and 1.6.7:
Try to compile this sketch for an AVR board. It results in a compilation error: _"'f2' was not declared in this scope"_. 

#ifdef __SAM3X8E__ // Arduino Due compatible
void fake()
{
}
#else
void fake () 
{
}
#endif

void f1(){ f2(); }
void f2(){;}

void setup() {
}
void loop() {
}

But when lines 3 and 4 are commented out like this:
#ifdef __SAM3X8E__ // Arduino Due compatible
void fake()
//{
//}
#else
...
the sketch compiles just fine. Why does a code that shouldn't be included prevents a compilation?

ffissore, 2015-11-09:
Confirm. Thank you @qbahn21

ffissore, 2015-11-09:
This issue was moved to arduino/arduino-builder#51

closed 2015-11-09 by ffissore