arduino/Arduino #4119: library manager isn't picking up Firmata 2.5.0

soundanalogous, 2015-11-09:
Not sure what I'm missing. Tag is semver compliant '2.5.0'. library.properties file is updated, etc. Had same issue when I released 2.4.4 as well.

tinkerspy, 2015-11-09:
I logged in to report a similar issue, see this forum post:
http://forum.arduino.cc/index.php?topic=358171.0

I tried and tried but couldn't get the library manager to pick up my library at all until I deleted the cached json file.

Some observations:
- When opening the library manager it *does* request the json list (checked with fiddler)
- When requesting the json list it updates the timestamp on the cached file
- The contents of the cached file is *not* updated until you manually delete it.

This bug means that new libraries and updated libraries are not picked up on all the miriads of existing Arduino IDE installs! Observed on version 1.6.6. on Windows 7.

Chris--A, 2015-11-10:
I concur, I posted my solution in that thread as the same issue was found when someone was trying to get the new lib for the Wifi 101, so its not just updated libraries.

----

I found this problem when migrating from 1.6.4 to 1.6.5, as the AppData/local folder was decided as the best spot, however the previous (roaming) Arduino15 folder had precedence over the local Aruduino15, and the code I had tweaked in the core and libs did not get used until I deleted the roaming folder.

I want to attempt a PR to add a 'Force refresh' button to the lib manager, however before I do, there might be a simpler bug fix in the code. I'd also recommend adding a check into the IDE startup code to remove any lingering roaming folders in windows (Only needs to be a temporary fix, maybe for a couple of versions).

ffissore, 2015-11-11:
The problem is that we are not fetching directly from your repository, but rather from our fork: https://github.com/arduino-libraries/Firmata @cmaglie should we reconsider our choice and fetch from @soundanalogous repo directly?

ffissore, 2015-11-11:
Done. From now on, Firmata repo used by our publishing job will be the one from @soundanalogous Thank you for your patience Jeff

closed 2015-11-11 by ffissore