arduino/Arduino #4120: ArduinoISP driver no recognize the hardware

csauvard, 2015-11-09:
if you download the driver (ArduinoISP_WindowsDrivers) on Arduino.cc the file
arduinoISP.inf content :
DeviceID   = "VID_2341&PID_0049"

even hardware can comes with the hardware "VID_2A03&PID_0049"

so change 2341 by 2A03 on this file and start again and it's magic !!!!!

matthijskooijman, 2015-11-09:
Do you mean the driver for this piece of hardware? https://www.arduino.cc/en/Main/ArduinoISP

I think 2a03 is the USB VID of arduino.org, not arduino.cc, so I think you should look there for the proper drivers (having said that, @cmaglie, perhaps the driver could be changed to include the .org vid?).

csauvard, 2015-11-09:
yes

oh why not but no driver found on Arduino.org

regards

----- Mail original -----

csauvard, 2015-11-10:
Dear,
OK Yes effectively it's .ORG
i haven't never see the difference between two 

I haven't found driver on .ORG for this ISParduino R3. !!

I have a error state with IDE 1.7.7 with the .CC driver modified by me because IDE search again the other VID before sending !!!!

regards

christophe

----- Mail original -----

cmaglie, 2015-11-10:
The .org driver is included, but it seems that the support for ArduinoISP is missing even there.
I'm closing this issue, you should really ask them to add the support in their official driver, we can't do it since the USB VID 0x2A03 is owned by .org.

closed 2015-11-10 by cmaglie

csauvard, 2015-11-10:
OK right,

thanks 

regards

----- Mail original -----