arduino/Arduino #4121: SD make83Name doesn't allow extended ASCII characters

mcaldwelva, 2015-11-09:
This line of code is overly restrictive:
>if (i > n || c < 0X21 || c > 0X7E)return false;

Windows will use extended ASCII in short file names if they are present in the long file name. Based on the Wikipedia entry for FAT 8.3 file naming, something like this seems more appropriate:
>if (i > n || c < 0x20 || c == 0x7F) return false;

I hope you will consider making a change along these lines in the future.  Thanks!

cmaglie, 2015-11-11:
@mcaldwelva 
looking at this article from Microsoft:

https://support.microsoft.com/en-us/kb/142982

it seems that spaces are not allowed in 8.3 so they should not be allowed (with `c < 0x20`).

Also I'm not sure that extended chars (>0x7F) are allowed, the article above leave this case undefined. Are you sure that it's possible? Does this patch solve a real world issue?

mcaldwelva, 2015-11-12:
Thanks for the quick response! Yes, this is a problem that anyone might
encounter. I discovered it when I realized that the SD library wasn't
finding several files on my card and narrowed it down to those with Latin
characters. Here are a couple examples of long names and the Windows
generated short names:

06. ¿Por Qué Te Vas_.mp3
06¿POR~1.MP3

05. Penélope.mp3
05PENÉ~1.MP3

Windows shouldn't generate short file names with spaces, but other systems
may. I believe space is allowed anywhere except for the first char, but
there is a separate line of code to handle that case.  Thanks,

.Mike

agdl, 2016-07-12:
This issue was moved to arduino-libraries/SD#3

closed 2016-07-12 by agdl