arduino/Arduino #4122: Arduino doesn't install silently.

Darathorn, 2015-11-10:
I want to uninstall arduino silently, by running the uninstall.exe, with the parameter /S, in commado prompt.
When I do this, it still prompt if I am sure that I want to do this, since it will delete everything in the arduino directory. I need it, to not do this, since it is not "silent". Or, there should be a way, so force uninstall, so it just does it, after being told, without any issues.

ffissore, 2015-11-11:
I'm sorry, we don't support silent uninstalls at the moment. We tried making silent Installs, but that failed too because Windows want to ask you permission for installing drivers, no matter what CLI params we use

erikng, 2016-01-19:
I may be able to help with the silent installation of drivers if this could resolve the issue with silent uninstalls.

Our organization can silently install the drivers using a script, but we will need silent uninstalls.

erikng, 2016-01-20:
@ffissore it looks like there are two issues with regards to installing drivers.

In order to silently install drivers, they have to be a trusted publisher.

You can do the following:

`certutil.exe -addstore -f "TrustedPublisher" Certificate.cer"`

Unfortunately in my testing with 1.6.7, your SHA2 certificate "Arduino SRL" is not allowing this to silently install on Windows 7 64-Bit. All drivers issued with "Arduino LLC" will install silently if the certificate is a trusted publisher.

I'm beginning to think that this is a bug in Windows 7 and I cannot seem to find much documentation on silently installing SHA2 signed drivers.

This is the closest information I can find:

https://social.msdn.microsoft.com/Forums/en-US/94546a3a-cba2-423d-8e11-7708623a65c6/unattended-driver-update-in-headless-mode-asks-for-always-trust-software-from-xxxx?forum=quebecservicingdeployment

http://stackoverflow.com/questions/33778515/dpinst-silent-signed-driver-installation-fails-on-windows-7

ffissore, 2016-01-20:
Hi @erikng, I'm sorry but you've missed the [news](https://groups.google.com/a/arduino.cc/forum/#!topic/developers/YJLX6AZHem4): I've left Arduino about a month ago. Please refer to @cmaglie from now on. Thank you :)