arduino/Arduino #4123: Function prototype generator ignores "static" keyword

nickgammon, 2015-11-10:
See forum post: http://forum.arduino.cc/index.php?topic=358602

Example code:

```c++
static void doSomething()
{

}

void setup() {
  doSomething();
}

void loop() {

}
```

Generates errors in IDE 1.6.6:

```
/tmp/arduino_06b29c2745641c533fb2062c298c350d/sketch_nov10a.ino: In function 'void doSomething()':
sketch_nov10a:2: error: 'void doSomething()' was declared 'extern' and later 'static' [-fpermissive]
 static void doSomething()
                         ^
sketch_nov10a:2: error: previous declaration of 'void doSomething()' [-fpermissive]
 static void doSomething()
      ^
exit status 1
'void doSomething()' was declared 'extern' and later 'static' [-fpermissive]
```

The user did not in fact declare something one way and later another way.

Generated code from the IDE:

```c++
#include <Arduino.h>
#line 1
#line 1 "/tmp/arduino_06b29c2745641c533fb2062c298c350d/sketch_nov10a.ino"

void doSomething();
void setup();
void loop();
#line 2
static void doSomething()
{

}

void setup() {
  doSomething();
}

void loop() {

}
```

The "static" keyword was dropped from the generated prototype.

---

It is perfectly valid to add "static" to a function definition, in order to stop that function name from clashing with functions named the same in other compilation units.

nickgammon, 2015-11-10:
This same code compiles OK in IDE 1.6.5.

It would be nice if you tested changes to the function prototype-generator to make sure that code does not break from one minor release of the IDE to another. Maybe have some regression testing suites?

In IDE 1.6.5 this is what the generated output from the IDE is:

```c++
#line 1 "sketch_nov10c.ino"
#include "Arduino.h"
static void doSomething();
void setup();
void loop();
#line 1
static void doSomething()
{

}

void setup() {
  doSomething();
}

void loop() {

}
```

It looks like someone deliberately removed the support for "static".

ffissore, 2015-11-10:
Thank you @nickgammon This was reported in #4079 and fixed. Fix is available with hourly builds and will become part of 1.6.7

closed 2015-11-10 by ffissore

malipetek, 2016-02-03:
Thanks people.