arduino/Arduino #4124: 1.65 Doesn't Launch Completely

mrmonteith, 2015-11-10:
I've been using older versions of the Arduino IDE.  But 1.65 launches, shows "Loading Configuration", and then closes with no error messages.  I've tried both the Windows installer and the unzipped version.  I've even removed software that I had updated since the version that was working.  Still no go.  I finally went back to version 1.64 and it works fine.  

 Here are the system details.
Win 7 Pro, SP1, i5-3470 @3.20Ghz, 8GB ram, 64-bit

NicoHood, 2015-11-10:
1.6.6 is the latest version. Please try that again.

mrmonteith, 2015-11-10:
Sorry. 1.66 does the same thing.  It launches, I see "Loading Configuration" and it closes.

matthijskooijman, 2015-11-10:
@mrmonteith Could you try running "arduino_debug.exe" from a console to see if any error is printed?

mrmonteith, 2015-11-10:
C:\Users\mmonteith\Desktop\Arduino Installs\arduino-nightly>arduino_debug
Loading configuration...
com.sun.jna.platform.win32.Win32Exception: The system cannot find the file specified.
        at com.sun.jna.platform.win32.Advapi32Util.registryGetStringValue(Advapi32Util.java:580)
        at com.sun.jna.platform.win32.Advapi32Util.registryGetStringValue(Advapi32Util.java:555)
        at processing.app.windows.Platform.recoverDefaultSketchbookFolder(Platform.java:66)
        at processing.app.windows.Platform.init(Platform.java:57)
        at processing.app.Base.guardedMain(Base.java:151)
        at processing.app.Base.main(Base.java:135)

I'm running Java 8 updated 65.  I even removed and reinstalled it.  Still have to have to go back to 1.6.4 to run.

matthijskooijman, 2015-11-10:
@ffissore, @cmaglie, I'll leave this one to you guys from here on :-)

ffissore, 2015-11-11:
It looks like your windows is missing a "Personal" key in windows registry, under `Software\Microsoft\Windows\CurrentVersion\Explorer\Shell Folders`. It should tell us where your "Documents" folder is located. Can you confirm that entry is missing? Any other entry that look like your Documents folder?

mrmonteith, 2015-11-11:
It looks like it might be missing.  But what's odd is that it works fine with the older versions and haven't had a problem until now.

ffissore, 2015-11-16:
Can you tell which windows registry key your windows uses to tell that a folder is your "Documents" folder?

mrmonteith, 2015-11-16:
Possibly a My Documents entry.  I tried both a Documents and a My Documents entry to see if it would work and still doesn't.    Why does the older versions work and the new ones don't.  

 With the debug it really doesn't tell you what it's having a problem with reading and what.  That would help a lot.  If I knew what it was looking for and where it expected it.

ffissore, 2015-11-18:
The code that's failing is [this one](https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/windows/Platform.java#L66-L66): the IDE is accessing your Windows registry, entry `Software\Microsoft\Windows\CurrentVersion\Explorer\Shell Folders`, key `Personal`, to discover where is your `My Documents` folder
I'm googling for why your windows should miss that entry

mrmonteith, 2015-11-18:
Well it's a work pc so it looks like some folders were pointing to a shared drive.  So maybe it was removed but never pointed to the shared drive.  In any case I created a String Value  entry called Personal and pointed it to my documents directory and it works fine.

ffissore, 2015-11-18:
@mrmonteith latest windows hourly build contains a fix (an attempt really). Can you help me debug this issue and try this build, possibly after removing the registry entry you manually created?

ffissore, 2015-11-18:
Hourly url is http://www.arduino.cc/en/Main/Software#hourly

mrmonteith, 2015-11-18:
I don't mind helping at all.  I downloaded the 18 November 2015 9:47:20 GMT.  I renamed the entry to oldPersonal so it wouldn't find it.  I get the following:
Loading configuration...
java.lang.IllegalStateException: Unable to find Personal key in Windows registry
        at processing.app.windows.Platform.readRegistryEntry(Platform.java:81)
        at processing.app.windows.Platform.recoverDefaultSketchbookFolder(Platform.java:67)
        at processing.app.windows.Platform.init(Platform.java:58)
        at processing.app.Base.guardedMain(Base.java:151)
        at processing.app.Base.main(Base.java:135)

ffissore, 2015-11-18:
So you also miss a User Shell Folders entry. Gosh. No, I don't think it's safe: this kind of stuff changes at every windows version. I'll keep on googling for a better way and keep you posted

ffissore, 2015-11-18:
Should you have a windows hacker at hand, take him/her to your desk pls :)

mrmonteith, 2015-11-18:
I'm as close as it gets. I've done windows since there was Windows.  Just I don't do a whole lot of that any more.  If there isn't a Personal entry you could default to Desktop.  Just depends on what the software is needing that folder for.  It might be that these domain admins rewrote some of the entries to point to the share and just forgot this one. If I had my home laptop and pc in front of me I could do some comparing.

ffissore, 2015-11-19:
Thank you @mrmonteith I will then ask to try some more fix attempts as I find them :)

ffissore, 2015-11-19:
I've just pushed a new hourly. This will first attempt to use windows registry: if that fails, will fallback to %USERPROFILE%\Documents, if that exists

mrmonteith, 2015-11-19:
That worked.  I downloaded the 19 November 2015 7:15:19 GMT.

ffissore, 2015-11-19:
Good! Closing the issue then. Thank you!

mrmonteith, 2015-11-19:
Awsome. Glad I could help.

teliot, 2015-11-27:
is my issue related to this patch? am getting errors on nightly, will need to double check if nightly error was same as the main dl.
http://forum.arduino.cc/index.php?topic=362134.0

ffissore, 2015-11-27:
yes @teliot you're having the same issue. 1.6.7 will soon be release. In the meanwhile, you can use the hourly build http://www.arduino.cc/en/Main/Software#hourly

teliot, 2015-11-27:
i reverted my quick fix and tested hourly build, IDE fails to load. please reopen bug :(

looking at the code my registry keys are in var findInUserShellFolders which is unused. the var that is being used will most likely be null in my windows 8.1 build 9600. 

related: http://stackoverflow.com/questions/585534/what-is-the-best-way-to-find-the-users-home-directory-in-java

which mentions shgetfolderpath that is deprecated after win xp, vista forward uses shgetknowngolderpath.
http://stackoverflow.com/questions/18493484/shgetfolderpath-deprecated-what-is-the-alternative
http://bugs.java.com/bugdatabase/view_bug.do?bug_id=4787931

teliot, 2015-11-27:
exported registry key info:

Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\Shell Folders]
"!Do not use this registry key"="Use the SHGetFolderPath or SHGetKnownFolderPath function instead"

Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\User Shell Folders]
"Desktop"=%USERPROFILE%\Desktop
"Local AppData"=%USERPROFILE%\AppData\Local
"Startup"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Startup
"Cookies"=%USERPROFILE%\AppData\Local\Microsoft\Windows\INetCookies
"SendTo"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\SendTo
"Personal"=%USERPROFILE%\Documents
"Recent"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Recent
"Favorites"=%USERPROFILE%\Favorites
"My Pictures"=%USERPROFILE%\Pictures
"Start Menu"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Start Menu
"NetHood"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Network Shortcuts
"My Music"=%USERPROFILE%\Music
"My Video"=%USERPROFILE%\Videos
"Cache"=%USERPROFILE%\AppData\Local\Microsoft\Windows\INetCache
"Programs"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Start Menu\Programs
"History"=%USERPROFILE%\AppData\Local\Microsoft\Windows\History
"{374DE290-123F-4565-9164-39C4925E467B}"=%USERPROFILE%\Downloads
"Templates"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Templates
"AppData"=%USERPROFILE%\AppData\Roaming
"PrintHood"=%USERPROFILE%\AppData\Roaming\Microsoft\Windows\Printer Shortcuts

if the user has roaming profiles in a domain environment depending on how it was setup the %USERPROFILE% environment var may or may not be used in the reg key.

ffissore, 2015-11-30:
@teliot please start the IDE using `arduino_debug.exe` and paste the error message it prints

teliot, 2015-12-01:
this is the error i was getting.

:\Program Files (x86)\Arduino>arduino_debug.exe
Loading configuration...
com.sun.jna.platform.win32.Win32Exception: The system cannot find the file speci
fied.
       at com.sun.jna.platform.win32.Advapi32Util.registryGetStringValue(Advapi
32Util.java:580)
       at com.sun.jna.platform.win32.Advapi32Util.registryGetStringValue(Advapi
32Util.java:555)
       at processing.app.windows.Platform.recoverSettingsFolderPath(Platform.ja
va:61)
       at processing.app.windows.Platform.init(Platform.java:56)
       at processing.app.Base.guardedMain(Base.java:151)
       at processing.app.Base.main(Base.java:135)

ffissore, 2015-12-01:
Did you unpack the hourly inside `C:\Program Files (x86)` ? If not, then you're using 1.6.6 to generate the error. 1.6.6 is affected by this bug: the fix is available only with hourly builds. Can you double check you're using the hourly?

stefanbr1990, 2015-12-01:
I'm having the same issue.

C:\arduino-nightly-windows\arduino-nightly\arduin
o_debug.exe
Loading configuration...
com.sun.jna.platform.win32.Win32Exception: Het systeem kan het opgegeven bestand
 niet vinden.
        at com.sun.jna.platform.win32.Advapi32Util.registryGetStringValue(Advapi
32Util.java:580)
        at com.sun.jna.platform.win32.Advapi32Util.registryGetStringValue(Advapi
32Util.java:555)
        at processing.app.windows.Platform.fixSettingsLocation(Platform.java:259
)
        at processing.app.PreferencesData.init(PreferencesData.java:36)
        at processing.app.BaseNoGui.initParameters(BaseNoGui.java:993)
        at processing.app.Base.guardedMain(Base.java:158)
        at processing.app.Base.main(Base.java:136)

I've looked into the registry and it has it's setting 
HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\User Shell Folders.

I'm not sure but for some reason 1.6.4 worked fine in our network. 

Any ideas?

ffissore, 2015-12-01:
Ahh yes, thank you. Fix coming

closed 2015-12-01 by ffissore

ffissore, 2015-12-01:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

teliot, 2015-12-01:
the need to unpack in program file's seems odd. ran the hourly from desktop and it worked fine with the System.getProperty("user.home") changes.

Thanks

commit 46d1c890 (2015-12-01): Windows: even old settings folder may be missing from the registry. Fixes #4124