arduino/Arduino #4125: Preprocessor logic in 1.6.6 changed causing compilation issues with existing sketches

arkhipenko, 2015-11-10:
It seems that the logic of creating function prototypes automatically has been changed, leading to compilation issues with existing sketches. 

For instance:
This will compile under 1.6.5 and fail to compile under 1.6.6:

		void (*p)() = &test;

		void setup() {
		  test();
		}

		void loop() {
		}

		void test() {
		}

This will compile under both 1.6.5 and 1.6.6:

		void setup() {
		  test();
		}
		void (*p)() = &test;

		void loop() {
		}

		void test() {
		}

This will also compile under both 1.6.5 and 1.6.6:

		void setup() {
		}
		void (*p)() = &test;

		void loop() {
		  test();
		}

		void test() {
		}

Seems like 1.6.6 pre-processor does not allow forward references ONLY at the beginning of the sketch, while 1.6.5 was OK with it.

per1234, 2015-11-10:
Try the hourly build: https://www.arduino.cc/en/Main/Software#hourly
I reported your issue(as I stated in the forum thread where you reported it) here: https://github.com/arduino/arduino-builder/issues/50 and ffissore has been working on it.
Unfortunately the examples from your [TaskScheduler](https://github.com/arkhipenko/TaskScheduler) library still don't compile. I discovered that in my testing last night and wanted to wait until https://github.com/arduino/arduino-builder/commit/40328a0a169b2d3cbc5f42348b134d4782f66ca9 had reached the hourly build to retest but it looks like that issue still exists.

arkhipenko, 2015-11-11:
Thanks. 
I have actually updated the examples with function prototypes, and am about to push into the repo. 
It was the right thing to do anyway.
I am more concerned that DirectIO library does not compile under 1.6.6 anymore, and I use that library A LOT. Trying to find the solution, but might end up reporting it here as well....

matthijskooijman, 2015-11-11:
> I am more concerned that DirectIO library does not compile under 1.6.6 anymore, and I use that library A LOT. Trying to find the solution, but might end up reporting it here as well....

If you can isolate the issue there, please report it, it helps us minimize the amount of breakage. If you do, best report it here: https://github.com/arduino/arduino-builder

closed 2015-11-11 by matthijskooijman