arduino/Arduino #4126: Conditional library include statements incorrectly cause that library to be linked in.

SlashDevin, 2015-11-10:
Consider this simplified example:

    #if 0
      #include <NeoHWSerial.h>
    #endif

    void setup()
    {
      Serial.begin( 9600 );
    }

    void loop() {}

This forces the NeoHWSerial library to be linked in, even when the condition is FALSE and there are no compile-time references to that library.

As a result, ISRs in HardwareSerial (via 'Serial') conflict with the same ISRs in the incorrectly linked-in NeoHWSerial.  Simply commenting out the NeoHWSerial include statement removes the "require" of that library, and HardwareSerial links in correctly, as the only source of the ISRs:

    #if 0
      //#include <NeoHWSerial.h>
    #endif

    void setup()
    {
      Serial.begin( 9600 );
    }

    void loop() {}

I found this on 1.0.5 and reproduced it on 1.6.5r2

matthijskooijman, 2015-11-10:
Could you try 1.6.6? I think this was solved in that version.

SlashDevin, 2015-11-11:
Yep, fixed.  Thanks!

closed 2015-11-11 by SlashDevin