arduino/Arduino #4128: Due Serial.flush() missing last byte

borisff, 2015-11-10:
http://forum.arduino.cc/index.php?topic=153928.0

To show the problem, add to
\sam\1.6.6\cores\arduino\HardwareSerial.h
public:
void disableTXRX() {};
same in
\sam\1.6.6\cores\arduino\UARTClass.h
In
\sam\1.6.6\cores\arduino\UARTClass.cpp
Add
void UARTClass::disableTXRX(void) {
_pUart->UART_CR = UART_CR_RSTRX | UART_CR_RXDIS | UART_CR_RSTTX | UART_CR_TXDIS;
}

Create a test sketch using Serial2&3, Serial 2 sends to Serial3, 9600 baud,
write 100 bytes in a loop using Serial2
than immediately call
Serial2.flush();
Serial2.disableTXRX();

There will be a	byte missing on Serial3 RX.
If you add a delay(5) just after flush(),
no missing byte.

The fix is in \sam\1.6.6\cores\arduino\UARTClass.cpp
in  void UARTClass::flush( void )
Replace as proposed by 'pylon'
while ((_pUart->UART_SR & UART_SR_TXRDY) != UART_SR_TXRDY);
by
while ((_pUart->UART_SR & UART_SR_TXEMPTY) != UART_SR_TXEMPTY);

and no need to add a delay after flush() to have no missing byte.

borisff, 2015-11-10:
[image]

sandeepmistry, 2015-11-17:
@borisff thank you for reporting the issue, steps to reproduce, and suggesting a patch.

I've confirmed the behaviour and created #4173 with a patch as per your suggestion.

I was testing using the following sketch, with ``TX2`` connected to ``RX3``:
```arduino
void setup() {
  Serial.begin(115200);

  Serial2.begin(9600);
  Serial3.begin(9600);

  for (int i = 0; i < 100; i++) {
    Serial2.write(i);
  }

  Serial2.flush();
  Serial2.disableTXRX();
}

void loop() {
  if(Serial3.available()) {
    int c = Serial3.read();

    Serial.println(c);
  }
}
```

and the following SAM core patch to disable TX and RX:
```
diff --git a/hardware/arduino/sam/cores/arduino/HardwareSerial.h b/hardware/arduino/sam/cores/arduino/HardwareSerial.h
index 5674e57..6921bea 100644
--- a/hardware/arduino/sam/cores/arduino/HardwareSerial.h
+++ b/hardware/arduino/sam/cores/arduino/HardwareSerial.h
@@ -35,6 +35,8 @@ class HardwareSerial : public Stream
     virtual size_t write(uint8_t) = 0;
     using Print::write; // pull in write(str) and write(buf, size) from Print
     virtual operator bool() = 0;
+
+    void disableTXRX() {};
 };
 
 extern void serialEventRun(void) __attribute__((weak));
diff --git a/hardware/arduino/sam/cores/arduino/UARTClass.cpp b/hardware/arduino/sam/cores/arduino/UARTClass.cpp
index ed1cb26..180fc0d 100644
--- a/hardware/arduino/sam/cores/arduino/UARTClass.cpp
+++ b/hardware/arduino/sam/cores/arduino/UARTClass.cpp
@@ -196,3 +196,7 @@ void UARTClass::IrqHandler( void )
   }
 }
 
+void UARTClass::disableTXRX(void) {
+  _pUart->UART_CR = UART_CR_RSTRX | UART_CR_RXDIS | UART_CR_RSTTX | UART_CR_TXDIS;
+}
+
diff --git a/hardware/arduino/sam/cores/arduino/UARTClass.h b/hardware/arduino/sam/cores/arduino/UARTClass.h
index 3747d8b..0b7de4d 100644
--- a/hardware/arduino/sam/cores/arduino/UARTClass.h
+++ b/hardware/arduino/sam/cores/arduino/UARTClass.h
@@ -62,6 +62,8 @@ class UARTClass : public HardwareSerial
 
     operator bool() { return true; }; // UART always active
 
+    void disableTXRX(void);
+
   protected:
     void init(const uint32_t dwBaudRate, const uint32_t config);
```

Without #4173 ``0`` to ``98`` would be printed with a 255 (-1), and with the patch ``0`` to ``99`` is printed as expected.

cmaglie, 2015-11-18:
Fixed with #4173, thank you!

closed 2015-11-18 by cmaglie