arduino/Arduino #4129: sprintf(ptr, "%x", 0xDB) returns "FFDB" on an UNO using 1.6.6

nutterts, 2015-11-11:
Reading documentation on sprintf and the char datatype leads me to believe this is a bug.

The same thing happends when using Serial.println(0xDB, HEX) 

It's not just 0xDB but every value > 127 which makes it obvious it's related to 'char' being an unsigned data type. 

But since we're converting it into it's hex. representation it seems like an unexpected result.

Attached ino file reproduces the problem. And sure enough, if you change "char chr 0xDB" to "unsigned char = 0xDB" it does work. 

[sprintf_bug.ino.txt](https://github.com/arduino/Arduino/files/31378/sprintf_bug.ino.txt)

NicoHood, 2015-11-11:
Uhm I am not sure if printf is supported. not all default std functions are available on arduino.
you might want to try this one: https://github.com/Chris--A/PrintEx

matthijskooijman, 2015-11-11:
`sprintf` should work, though, it prints to a buffer.

I suspect that your 0xDB is interpreted as a signed char, and then sign-extended to a signed int. However, integer literals should always become at least of size int (and wouldn't even fit inside signed char, so it should be at least unsigned char to make it fit).

However, testing `Serial.println(0xDB, HEX)` works fine for me, printing "DB". Are you actually using that line code literally? Or are you doing:

    char foo = 0xDB;
    Serial.println(foo, HEX);

In the latter case, printing FFDB is expected. You assign 0xDB to a (signed) char variable. Since that doesn't fit, the value overflows and is interpreted as -37 instead. Now, when you pass `foo` to `Serial.println`, it is integer promoted to `int`, which is 16 bits. To keep the correct value, it is sign-extended. In other words, -37 stored as an int is 0xFFDB. Then, the value is converted to an unsigned number and printed, resulting in your FFDB print.

So I suspect this behaves as expected. If you want to store 0xDB in a variable, you will at least need to make it an `unsigned char` variable.

cmaglie, 2015-11-11:
The explanation of @matthijskooijman is correct, this is not an Arduino / compiler bug but the expected result (as weird as it may seems).

You can convince yourself by trying the following example on a PC (I tried on linux):

```C++
#include <stdio.h>
 
int main()
{
  char str[20];
  char chr = 0xDB;
  sprintf(str, "%x", chr);
  printf("%s\n", str);
  return 0;
}
```

that prints `ffffffdb` (in this case the int size is 32 bit, that's the reason for the 4 extra `f`).

closed 2015-11-11 by cmaglie

nutterts, 2015-11-11:
Yeah, it's not what I would expect when converting to it's hex representation. But hey, if it's expected behaviour then it's not a bug but a feature. :)

@cmagie, @matthijskooijman Thx for clearing it up. Didn't have my linux machine handy yesterday but I checked and you both are totally right. 

Strange that I never noticed it before and I really looked online and in my books to see if it was expected behauviour before posting. O well, never to old to learn.