arduino/Arduino #4130: [Boards Manager]Unable to reinstall Arduino AVR Boards 1.6.9 after installing previous version

per1234, 2015-11-11:
Using Arduino IDE 1.6.7 2011-11-10 12:55 with Windows 7 64 bit

- Tools > Board > Boards Manager
- Arduino AVR Boards > Select Version... 1.6.8 > Install
  - 1.6.8 is installed correctly.
- Close
  - the issue doesn't occur if you don't close Boards Manager after installing 1.6.8.
- Tools > Board > Boards Manager
- Arduino AVR Boards > Upgrade
  - 1.6.8 is deleted from C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\avr and all files are deleted from C:\Program Files (x86)\arduino-nightly\hardware\tools.

Now when you try to compile anything you get:
`exec: "/bin/avr-g++": file does not exist`

The same issue occurs with Arduino IDE 1.6.6.

NicoHood, 2015-11-11:
> Using Arduino IDE 1.6.7 2011-11-10 12:55 with Windows 7 64 bit

2011?

per1234, 2015-11-11:
Fixed, thanks!

per1234, 2017-07-04:
Boards Manager no longer has the bad behavior of deleting things in the Arduino IDE installation folder. Closing this as solved.

closed 2017-07-04 by per1234