arduino/Arduino #4132: Libraries and other data files don't belong in "Program Files" folders

ffissore, 2015-11-11:
_From @MikeyMoMo on November 11, 2015 0:51_

PLEASE get the libraries out of Program Files.  Windows 10 does not take it well when changes
are made to Program Files.  I am able to put files in and not be able to see them with Files
Explorer but can see them with a text editor.  Even after a reboot, I can edit the files but
not see them in Explorer and the IDE cannot find them.  This is a Windows design feature 
but you are exacerbating it by using Program Files for data storage.  Library code is data.  It can change.  When debugging a library, it can change frequently.  Windows does not handle this well.
This action is against the philosophy of Windows re the Program Files folders.  They are for code, not data.

_Copied from original issue: arduino/arduino-builder#54_

ffissore, 2015-11-11:
_From @matthijskooijman on November 11, 2015 7:4_

If you need to add libraries, you should add them to your sketchbook, in the libraries subdirectory (not sure where that is stored by default on Windows, but it is shown in the preferences). If you need to modify a built-in library, copy it into your sketchbook and modify it there, the copy in your sketchbook takes precedence.

I don't think that supplying builtin libraries and other code in program files is a bad idea, those files are not *meant* to be changed. If you insist that they are moved out, where do you suggest they should be stored? And how would you suggest that upgrades are handled if you modified a library?

ffissore, 2015-11-11:
Please @MikeyMoMo tell us your thoughts. Your issue is about and old and well established feature and behaviour of the IDE: you're probably new to it and/or have read old (and wrong) documentation
As @matthijskooijman said, libs in program files are NOT meant to be changed. If you want to customized them, copy them your sketchbook libraries

lmihalkovic, 2015-11-11:
IMHO it makes sense to be able to support #4050. The benefit of doing it is that the changes required to the IDE to support it are next to nil, while opening a lot more doors (basically the feature comes almost for free after the editor code is cleaned up a bit).

closed 2015-11-16 by ffissore