arduino/Arduino #4135: Triple-click highlighting issue--Mac v1.6.6

Jed-SWSDT, 2015-11-11:
Triple-click mouse line highlighting functionality is broken in 1.6.6

In prior versions the triple-click appears to have been detected on mouse button_down, which made it possible to highlight an entire line, keep the button down and highlight whole sections without having to carefully position the cursor. This is the standard behavior in other Mac programs.

Now it appears that mouse button_release is used, making it impossible to do this. Triple-click does highlight the line, but it can now only be used for one line. If you triple-click on a variable name, it wants to drag the text, not select the line. Using four-clicks does not fix the issue as it then wants to drag a whole line of text.

Jed-SWSDT, 2015-11-19:
What type of help is needed for this? It is a 'feature' that is driving me nuts with the IDE to the point I'm ready to abandon it for Eclipse.

ffissore, 2015-11-20:
We need a macosx long time user with java skills. He/She needs to hack both the arduino IDE or rsyntaxtextarea (the editor component) and add/modify proper macosx keybindings. He/She must also find the right balance among macosx user expectations, which are **not** unvarying as one might expect

chrstphrchvz, 2016-03-20:
Just another user here…

Upstream issue: bobbylight/RSyntaxTextArea#59

To clarify for myself, the feature (triple-click-hold-and-drag) selects entire "paragraphs" in normal text, whereas in programming entire lines are selected, because each is separated by line breaks.

As for other Java UIs: JEditTextArea, which is what Processing appears to use at the moment, supports this feature; not sure how Eclipse supports it.