arduino/Arduino #4137: 1.6.6 no longer works with third-party boards/libraries

tigoe, 2015-11-12:
I;m finding that a few third-party boards and libraries that worked with 1.6.5 no longer work with 1.6.6.

Specifically I tested with the ESP8266 boards files from https://github.com/esp8266/arduino and the Attiny files from http://highlowtech.org/?p=1695. Both of these worked with 1.6.5.  It may be true for other third party boards, I haven't had time to test yet.

Process to produce the error:

1. Add the boards files for the two boards mentioned above, using the instructions provided and the Boards manager (e.g. add http://arduino.esp8266.com/stable/package_esp8266com_index.json and https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json to the Additional Board Manager URL field in Preferences)
2. Choose the appropriate board (Adafruit HuzzahESP8266 and Attint84, 8MHz Internal were my two choices)
3. Compile an example for the board in question. The error I get looks like this:

```
Board huzzah (platform esp8266, package esp8266) is unknown
Error compiling
```
Similar error on the ATTIny.

I also tried uninstalling and reinstalling the boards through the board manager, quitting, and restarting.  Same results.

Will try with other external boards and report as soon as I get a chance.

damellis, 2015-11-12:
Hmm... the ATtiny boards are working for me in 1.6.6. Which version of the boards do you have installed? (I have 1.0.1.) You don't have any old versions in your sketchbook folder, do you?

tigoe, 2015-11-12:
I was using yoru latest. Worked great in 1.6.5, not 1.6.6. Will try re-install when I get a break.

ffissore, 2015-11-17:
I've also tried compiling the empty sketch with both cores and it compiles without any issues

ladyada, 2015-11-21:
i def just tried huzzah/esp8266 on a new 1.6.6 win7 install and it installs ok.

tigoe, 2015-11-21:
Did it compile OK as well?  Installs fine on OSX, but doesn't compile.

tigoe, 2015-11-21:
Found the issue. This error helped me find it:

```
/Users/tigoe/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266: /Users/tigoe/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266
```
Looks like uninstalling then reinstalling without restarting causes a problem. I manually removed the whole /Users/tigoe/Library/Arduino15/packages/esp8266 dir, then reinstalled, and everything is fine now.

ladyada, 2015-11-21:
compiliation is sometimes-broken due to arduino-builder snagging on the ESP8266WiFi.h file and not generating function prototypes, might be related to people's difficulties. see https://github.com/esp8266/Arduino/issues/1066 for more deets

igrr, 2015-11-21:
Also see notes here: https://github.com/arduino/arduino-builder/issues/68
Looks like there are some issues with arduino-builder when c++11 stuff gets included from sketch (even indirectly as in ESP8266WiFi.h).

tigoe, 2015-11-22:
Yeah, seems to be the same issue as arduino/arduino-builder#68, thanks.

ffissore, 2015-11-23:
Bug being tracked at https://github.com/arduino/arduino-builder/issues/68. Closing

closed 2015-11-23 by ffissore

MikeAshby, 2015-11-27:
Try compiling something else like Lilypad Arduino. Find something that works, and then switch back to Adafruit Huzzah esp8266 and it works.

tigoe, 2015-12-11:
Bed news, it seems to be back again with the latest build of the ESP platform. Now the board is recognized, but implicit header declaration is not working. Thanks @lgrr for making the connection.