arduino/Arduino #4139: [Library Manager] Please add HardwareSerialRS485 release 2.1.0

MichaelJonker, 2015-11-12:
Hello, I made a new release (2.1.0) of the library HardwareSerialRS485

https://github.com/MichaelJonker/HardwareSerialRS485

I understood from your documentation that the Librarian would scan every hour for updates. Apparently this is maybe not the case. A previous update, release 2.0.1, to fix the URL to the documentation), was never picked up by the librarian.

Please let me know if I should do anything special, to make new releases automatically available through the Library Manager, or if I should each time open an issue.

Kind regards

Michael Jonker

cmaglie, 2015-11-13:
Hi @MichaelJonker,
I confirm the wrong behaviour of the IDE, for some reason it's picking the description of the first library listed (and not the latest) but only after installing the libary.

MichaelJonker, 2015-11-13:
Hi, as information:
in my IDE, not only the description is of the first library, also if I install the library I get the first version.
and I also see the wrong URL before the library is installed.
Maybe for some reason the library manager is 'blind' for the new versions...

Thanks for looking into this.
M

MichaelJonker, 2015-11-15:
Hello

some further information which may help:

I see that the file .../appData/local/Arduino15/library_index.json.tmp.gz is not refreshed when I open the library manager.
The file appData/local/Arduino15/library_index.json is refreshed, but as it is extracted from the library_index.json.tmp.gz file this does not update the library index.

If I delete the file library_index.json.tmp.gz, the correct new file is loaded and I can see the various versions of the libraries.
I think this implies that I will not see any new library after the first time the Library Manager loads the library_index.json.tmp.gz file....

I am using IDE 1.6.6 on windows 7

Hope this is useful information.

Kind regards, Michael Jonker

ffissore, 2015-11-17:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-17 by ffissore

MichaelJonker, 2015-11-22:
Great, thanks!

commit 1328f3a9 (2015-11-17): Library and Boards Managers were using old copies of library/boards indeces. thus making the UI show old data. Fixes #4139 #3904 #3795