arduino/Arduino #4141: Document / specify core and variant requirements and workings

matthijskooijman, 2015-11-13:
We need some more exhaustive documentation about the "interface" that a core or variant should expose. There is [this wiki page](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification), which documents a good deal of the structure and has examples, but is not exhaustive or explicit. In particular:
 - All of the required keys in platform.txt, boards.txt and programmers.txt, and what they are used for, should be listed. These keys form the "external interface" of the core, which can be used by the IDE or arduino-builder. A core can also contain arbitrary other keys, which are used for variable substitutions internally, but cannot be relied on to exist is all cores. Currently, this distinction is blurry and ad-hoc.
 - All of the macros, functions and variables that a variant of the Arduino core should expose should be documented. Some of these are mean to be used by the sketch directly (and should actually be documented as part of the normal Arduino reference), but some of them are only used internally by the Arduino core as part of another API. Currenly, no definitive list of these exists, making variants (both third-party and Arduino ones) inconsistent or incomplete.

Both of these pieces of documentation should be updated when the requirements change in future IDE or core versions, with the changes mentioned explicitly and with the IDE or core version that introduced them.

Related issues are #3371 and #3681.

matthijskooijman, 2015-11-13:
One  additional angle should be covered: The interface that the Arduino core offers to its variants and derived (board-only) platforms. E.g., what variables does each tool, recipe, programmer, etc. assume are present and how they are used.

per1234, 2015-11-13:
Other thing that should be documented(assuming it's the intended behavior) is that Burn Bootloader uses the platform.txt of the selected board but Upload Using Programmer uses the platform.txt of the selected programmer. This has caused confusion for me and at least two other 3rd party cores because it wasn't documented(see https://github.com/arduino/Arduino/issues/2886 and https://github.com/SpenceKonde/ATTinyCore/commit/9c501518037cfa78242500e5749e15ccce3418c8).

matthijskooijman, 2020-09-17:
Also note this proposal I made (quite) a while ago already, which suggests to apply some versioning scheme to these interfaces, to make it easier to change them in the future while keeping compatibility: https://groups.google.com/a/arduino.cc/g/developers/c/H6vCRlgRTCo/m/22oQfVY3AgAJ